OPCFoundation/UA-.NETStandard #2529: MonitoredItem Notification not triggered intermittently

bbmmcodegirl, 2024-02-20:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [X] Help wanted

### Current Behavior

We are using this OPC/UA implementation inside a process within a production environment.
Very rarely, we observe the Notification event for a `MonitoredItem` not being called, even though its value changed.
This causes our production process to pause, so it is quite crucial for us.
This is intermittent. We may receive 20 notifications for value changes, but then one of the notifications is missing, so we miss out on the change of that value.
We currently use version 1.4.371.91 of the package, but this has also been observed with previous versions.
The occurrence is very rare, so it is very hard to establish what the specific conditions are that make this occur.
Note that when we had it occur, the value in the OPC/UA server was correct, as verified with a different tool. Only the `Notification` event of the associated `MonitoredItem` was not raised when the value had changed.

### Expected Behavior

Our `<OnNotification>` event handler should be called whenever the value of the monitored item changes.

### Steps To Reproduce

Here is the code that we are using to subscribe.
```
// Create the endpoint
  EndpointDescription endpointDescription = CoreClientUtils.SelectEndpoint(<some configuration>, <serverAddress>, useSecurity: false);
  EndpointConfiguration endpointConfiguration = EndpointConfiguration.Create(<some configuration>);
  ConfiguredEndpoint endpoint = new ConfiguredEndpoint(null, endpointDescription, endpointConfiguration);

  _session = await Session.Create(
    <some configuration>,
    endpoint,
    updateBeforeConnect: false,
    checkDomain: false,
    <some application name>,
    30 * 60 * 1000,
    new UserIdentity(<my user>, <my passwrod>),
    preferredLocales: null
   ###);
 _session.KeepAliveInterval = 5000;

 // Create a subscription for receiving data change notifications
 _logger.Info("Subscribing to items {items}", <some description of the items>);

 // Define Subscription parameters
 Subscription subscription = new Subscription(_session.DefaultSubscription)
 {
    DisplayName = "<My Subscription>",
    PublishingEnabled = true,
    PublishingInterval = 100,
    DisableMonitoredItemCache = true,
 };

 _session.AddSubscription(subscription);

 // Create the subscription on Server side
 subscription.Create();
 _logger.Debug("New Subscription created with SubscriptionId = {SubscriptionId} for items {items}", subscription.Id, <some description of the items>);

  // Create MonitoredItems for data changes
  foreach (var item in variablesToMonitor)
  {
     MonitoredItem monitoredItem = new MonitoredItem(subscription.DefaultItem)
     {
         StartNodeId = <some node Id>(item),
         AttributeId = Attributes.Value,
         DisplayName = <some display name>(item),

         SamplingInterval = (int)Math.Round(item.PollFrequency.TotalMilliseconds)
     };
     _logger.Trace("Subscribing to {node}", monitoredItem.StartNodeId);

     monitoredItem.Notification += OnNotification;

     subscription.AddItem(monitoredItem);
  }
 
  // Create the monitored items on Server side
 subscription.ApplyChanges();

void OnNotification(MonitoredItem sender, IEncodable e)
{
      _logger.Trace($"Value\"{sender.DisplayName}\" changed");
      ... // other processing
}
```

The steps we currently use are:
1. Compile the code
2.  Install in the production environment
3. Run for several weeks independently
4. Wait until it occurs after 5, or 7, or 9 weeks.

### Environment

```markdown
- OS: Microsoft Windows 10 Enterprise LTSC 10.0.17763 Build 17763
- Environment: Windows Service running as Local System
- Runtime: .Net 4.8
- Nuget Version: 1.4.371.91
- Component: OPCFoundation.NetStandard.Opc.Ua.Client
- Server: CODESYS Gateway -x64 version 3.5.18.30
- Client: In-house built proprietary client application (a Windows Service)
```

mozerlou, 2024-02-27:
Hey! I am getting a similar issue on my end. Usually for me its the same nodes 
I have the following specifcations: 
OPC UA Client: Workstation.UaClient 3.11
Server: Basic OPC Server

I am getting the notifications for all other monitored items except a couple specific ones from time to time. Not too sure what is happening

bbmmcodegirl, 2024-05-29:
@mregen sorry for the late answer. It turns out that yes, it seems to usually be a notification for one particular node value that gets missed at these very irregular intervals. But that is probably due to the fact that this one node is the one that changes most often. It gets many more changes than all other nodes. It is the value that changes in 42% of all of the notifications we receive.

romanett, 2025-07-18:
@bbmmcodegirl I will close this, please reopen if it still occurs on the newest version & you have a scenario with logs of Server & Client.