OPCFoundation/UA-.NETStandard #2542: Only ask for supported ServerCapabilities

KircMax, 2024-02-29:
### Type of issue

- [ ] Bug
- [X] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

Currently the Client will request the Servercapabilities regardless of wether the server supports these profiles.
E.g.: 
Server_ServerCapabilities_OperationLimits_MaxNodesPerHistoryReadData,12165,Variable
Server_ServerCapabilities_OperationLimits_MaxNodesPerHistoryReadEvents,12166,Variable
Server_ServerCapabilities_OperationLimits_MaxNodesPerHistoryUpdateData,12167,Variable
Server_ServerCapabilities_OperationLimits_MaxNodesPerHistoryUpdateEvents,12168,Variable

Server_ServerCapabilities_OperationLimits_MaxNodesPerNodeManagement,11713,Variable

Now that by itself is not an issue - since the client also handles failure of accessing the values by just using the default configurations limits...
BUT what might happen by that is that a server might detect that the client tried to access inexistant nodes and e.g. log a warning that a client has misused the service read e.g. like so:
[image]

The only thing I could imagine that would prohibit the client from accessing these -nonexistant but well defined - nodes 
is for the client to check for the supported profile array and only if it does support historyreaddata access the limit for the historyreaddata
[image]

First of all I can live with the current behaviour, I just noticed that that is something that might be problematic (like described above)... 
What do you think? 

### Expected Behavior

Only supported Servercapabilities are requested.

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

KircMax, 2024-02-29:
just as a hint - the spec. does not allow the server to provide the node with value 0 (to indicate unsupported):
[image]

mregen, 2024-02-29:
hi @KircMax , this way it is faster for the client to read all known operation limits, otherwise an extra browse were necessary. The way it is defined in the spec I would expect clients to try a read and rather not log it in the server.

KircMax, 2024-03-01:
Hi @mregen thanks a lot for your answer and your perspective on this.

closed 2024-03-01 by KircMax