OPCFoundation/UA-.NETStandard #2550: Create CSR from UA Expert fails for ECC certificates

mregen, 2024-03-11:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

 ```
Opc.Ua.Core.dll!Opc.Ua.CertificateFactory.CreateSigningRequest(System.Security.Cryptography.X509Certificates.X509Certificate2 certificate, System.Collections.Generic.IList<string> domainNames) Line 312
	at D:\Source\Repos\UA-.NETStandard5\Stack\Opc.Ua.Core\Security\Certificates\CertificateFactory.cs(312)

System.ArgumentNullException
  HResult=0x80004003
  Message=Value cannot be null. (Parameter 'key')
  Source=System.Private.CoreLib
  StackTrace:
   at System.ArgumentNullException.Throw(String paramName)
   at System.ArgumentNullException.ThrowIfNull(Object argument, String paramName)
   at System.Security.Cryptography.X509Certificates.CertificateRequest..ctor(X500DistinguishedName subjectName, RSA key, HashAlgorithmName hashAlgorithm, RSASignaturePadding padding)
   at Opc.Ua.CertificateFactory.CreateSigningRequest(X509Certificate2 certificate, IList`1 domainNames) in D:\Source\Repos\UA-.NETStandard5\Stack\Opc.Ua.Core\Security\Certificates\CertificateFactory.cs:line 312

```

### Expected Behavior

CSR is returned to UA Expert

### Steps To Reproduce

Connect to ECC enabled ref server with sysadmin credentials, export CSR, null pointer exception

### Environment

```markdown
- OS: Windows
- Environment:
- Runtime:
- Nuget Version: 1.5.375 preview
- Component:
- Server:
- Client:
```

romanett, 2024-03-29:
Hi mregen, what settings did you specifiy, for me it is not reporoducable with current ecc branch (console server), and these input parameters: 
[image]

romanett, 2024-03-31:
fixed with: https://github.com/OPCFoundation/UA-.NETStandard/pull/2398/commits/3306aa91e8664fb4ee447f05baf8085f63ff59d2

#2398

closed 2024-03-31 by romanett