OPCFoundation/UA-.NETStandard #2552: IOP: ECC preview server throws BadCertificateRevocationUnknown when calling ApplyChanges after cert update

mregen, 2024-03-12:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

If strict revocation check is used, GDS Push cert update fails the `UpdateCertificate` --> `ApplyChanges` because there is no way to transmit a CRL with the updated certificate. Only the Root CA cert is transmitted.
Server needs to be manually restarted or crashes.

```
System.AggregateException
  HResult=0x80131500
  Message=One or more errors occurred. (BadCertificateRevocationUnknown)
  Source=System.Private.CoreLib
  StackTrace:
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at Opc.Ua.Bindings.TcpTransportListener.CertificateUpdate(ICertificateValidator validator, CertificateTypesProvider certificateTypesProvider) in D:\Source\Repos\UA-.NETStandard5\Stack\Opc.Ua.Core\Stack\Tcp\TcpTransportListener.cs:line 447

  This exception was originally thrown at this call stack:
    Opc.Ua.CertificateValidator.GetIssuer(System.Security.Cryptography.X509Certificates.X509Certificate2, Opc.Ua.CertificateIdentifierCollection, Opc.Ua.CertificateStoreIdentifier, bool) in CertificateValidator.cs
    Opc.Ua.CertificateValidator.GetIssuersNoExceptionsOnGetIssuer(System.Security.Cryptography.X509Certificates.X509Certificate2Collection, System.Collections.Generic.List<Opc.Ua.CertificateIdentifier>, System.Collections.Generic.Dictionary<System.Security.Cryptography.X509Certificates.X509Certificate2, Opc.Ua.ServiceResultException>) in CertificateValidator.cs
    Opc.Ua.CertificateTypesProvider.LoadCertificateChainAsync(System.Security.Cryptography.X509Certificates.X509Certificate2) in CertificateIdentifier.cs
    Opc.Ua.CertificateTypesProvider.LoadCertificateChainRawAsync(System.Security.Cryptography.X509Certificates.X509Certificate2) in CertificateIdentifier.cs

Inner Exception 1:
ServiceResultException: BadCertificateRevocationUnknown

```

### Expected Behavior

The revocation check should be releaxed for the cert update check, to allow to continue to run.
Later with GDS server puish the trustlists can be updated.

### Steps To Reproduce

Start GDS Server.
Start Winforms GDS Client.
Connect to UA ref server, register and update certificate using the known workflow.
--> see exception thrown.

### Environment

```markdown
- OS: Windows
- Environment: 
- Runtime: .NET 8.0
- Nuget Version: 1.5.375 preview
- Component:
- Server:
- Client:
```

romanett, 2024-03-12:
@mregen  we could create our own certificate validator for the Apply changes, same as with the UpdateCertificate Method (with setting m_rejectUnknownRevocationStatus = false):

```C#
                        CertificateValidator certValidator = new CertificateValidator();
                        CertificateTrustList issuerStore = new CertificateTrustList();
                        CertificateIdentifierCollection issuerCollection = new CertificateIdentifierCollection();
                        foreach (var issuerCert in newIssuerCollection)
                        {
                            issuerCollection.Add(new CertificateIdentifier(issuerCert));
                        }
                        issuerStore.TrustedCertificates = issuerCollection;
                        certValidator.Update(issuerStore, issuerStore, null);
                        certValidator.Validate(newCert);
```

https://github.com/OPCFoundation/UA-.NETStandard/blob/3f29a72a37f231ab5ab0de151a197169940b4159/Libraries/Opc.Ua.Server/Configuration/ConfigurationNodeManager.cs#L412C24-L421C57

Should I prepare a PR?

romanett, 2024-03-13:
Here is the issue discussed in mantis: https://mantis.opcfoundation.org/view.php?id=9328

mregen, 2024-03-14:
@romanett, now I remember the discussion. I think first of all the Winforms GDS client need to be fixed to update the Trustlist first. Then we need to check if it is ok to use an alternate validator with revocation check disabled.

romanett, 2024-03-15:
Fix provided: https://github.com/OPCFoundation/UA-.NETStandard-Samples/pull/596