OPCFoundation/UA-.NETStandard #2559: Do not remove support for .NET Standard 2.0 until really necessary

OPCLabs, 2024-03-17:
### Type of issue

- [ ] Bug
- [X] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

It came to my attention that the support for .NET Standard 2.0 might be removed from this project soon. 

I have a library built on top of UA-.NETStandard, and also an accompanying project that has a bunch of Roslyn analyzers. Consequently, the Roslyn analyzers need to reference UA-.NETStandard. The Roslyn analyzers must target .NET Standard 2.0 for them to be universally usable (see https://learn.microsoft.com/en-us/dotnet/csharp/roslyn-sdk/tutorials/how-to-write-csharp-analyzer-code-fix ). If support for .NET Standard 2.0 was removed from UA-.NETStandard, I would not be able to build the Roslyn analyzers anymore.

### Expected Behavior

Keep the .NET Standard 2.0 targeting as long as technically feasible. Note that, at least for my usage, there is actually no need for the real "OPC UA" code to run correctly (connections etc.). All that is needed is the ability to keep referencing UA-.NETStandard in projects that are targeting .NET Standard 2.0.

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

mregen, 2024-03-26:
Hi @OPCLabs , in order to build the analyzers, would it be sufficient to provide the Opc.Ua.Core library with netstandard2.0 support only? 
The concern is somebody can ship an app using client/server only with the netstandard2.0 libraries, which have some known limitations. But if only Core is needed, there is no risk that backward compatibility must be maintained.
Otherwise we need to maintain the client/server code for netstandard2.0.

OPCLabs, 2024-03-27:
Hi @mregen. If just Opc.Ua.Core library is kept with netstandard2.0, I would still need to restructure the solution significantly in order to get rid of the other package(s) for the purpose of building analyzers - because I also use the other packages. But, perhaps with just Opc.Ua.Core kept with netstandard2.0, the effort needed for the restructure would be lower. So unless you can/are willing to keep netstandard2.0 in other packages as well, keeping it in Opc.Ua.Core is at least a help.

marcschier, 2025-09-19:
@OPCLabs what in the OPC. UA. Core library do you need in your analyzer?  Just types and generated, no transport and config?  We want to deprecate . Net standard and enable ECC across the board, so what we need to know is what needs to remain . Net standard 2.0 for you to migrate to 378.  Please let us know., thanks! 

Also, if you want to contribute in the analyzer and source gen area reach out to me or Randy.

OPCLabs, 2025-09-22:
It is an intricate issue. The answer to "what do I need in the analyzer" is "everything" and "nothing" and the same time. Let me explain:

I have assemblies that depend on UA-.NET Standard assemblies. There are literally thousands usages of types/members from UA-.NETStandard in my libraries. The analyzer then references these upper layer libraries. So it needs "everything" in order to be loaded. But it does not actually have to execute any code from UA-.NETStandard. So it needs "nothing".

Anybody who will need to write analyzer that in any way, directly or indirectly, references UA-.NETStandard, will face the same issue (at least until Microsoft allows the analyzers to target .NET Framework or .NET instead of .NET Standard API).

marcschier, 2025-11-19:
@OPCLabs , we have decided to remove .netstd2.0 in 378 release from Opc.Ua.Core and all other libraries, except for the new Opc.Ua.Types library, which we will use in the model compiler and future source generation libraries (post 378).  

Opc.Ua.Types library does not contain any generated code, but basic built in types, node and node state system.  

With the help of the soon to be added source generator libraries which also will support .net std 2.0 you can generate any missing code if you need it for your analyzer.  

As mentioned, if you built an analyzer on this project that is useful for users of this project, please consider contributing it to this project and we can see how to fit it into the picture, reach out to me directly to discuss.

PR #2570 (2024-04-03): Reenable .NET Standard 2.0 builds for core library
## Proposed changes

For some specific applications like analyzers it is preferable to have a .NET Standard 2.0 build of the core library.
Client/server build for NetStandard 2.0 will be maintained, but is unsupported in Nuget package and is not part of the test matrix. 

## Related Issues

- Fixes #2559 

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fill these out after creating the PR._

- [ ] Bugfix (non-breaking change which fixes an issue)
- [x] Enhancement (non-bre [... cut]