OPCFoundation/UA-.NETStandard #2568: Sometimes Nuget debug build fails

mregen, 2024-03-28:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

```
  Zipping directory "D:\a\1\s\Stack\Opc.Ua.Core\obj\zipnodeset2" to "D:\a\1\s\Stack\Opc.Ua.Core\Schema\Opc.Ua.NodeSet2.xml.zip".
##[error]Stack\Opc.Ua.Core\Opc.Ua.Core.csproj(100,5): Error MSB3943: Failed to zip directory "D:\a\1\s\Stack\Opc.Ua.Core\obj\zipnodeset2" to file "D:\a\1\s\Stack\Opc.Ua.Core\Schema\Opc.Ua.NodeSet2.xml.zip".  The process cannot access the file 'D:\a\1\s\Stack\Opc.Ua.Core\Schema\Opc.Ua.NodeSet2.xml.zip' because it is being used by another process.
D:\a\1\s\Stack\Opc.Ua.Core\Opc.Ua.Core.csproj(100,5): error MSB3943: Failed to zip directory "D:\a\1\s\Stack\Opc.Ua.Core\obj\zipnodeset2" to file "D:\a\1\s\Stack\Opc.Ua.Core\Schema\Opc.Ua.NodeSet2.xml.zip".  The process cannot access the file 'D:\a\1\s\Stack\Opc.Ua.Core\Schema\Opc.Ua.NodeSet2.xml.zip' because it is being used by another process. [TargetFramework=net6.0]
  Zip Schema/Opc.Ua.NodeSet2.xml in obj\/zipnodeset2.
##[error]Stack\Opc.Ua.Core\Opc.Ua.Core.csproj(100,5): Error MSB3943: Failed to zip directory "D:\a\1\s\Stack\Opc.Ua.Core\obj\zipnodeset2" to file "D:\a\1\s\Stack\Opc.Ua.Core\Schema\Opc.Ua.NodeSet2.xml.zip".  The process cannot access the file 'D:\a\1\s\Stack\Opc.Ua.Core\Schema\Opc.Ua.NodeSet2.xml.zip' because it is being used by another process.
D:\a\1\s\Stack\Opc.Ua.Core\Opc.Ua.Core.csproj(100,5): error MSB3943: Failed to zip directory "D:\a\1\s\Stack\Opc.Ua.Core\obj\zipnodeset2" to file "D:\a\1\s\Stack\Opc.Ua.Core\Schema\Opc.Ua.NodeSet2.xml.zip".  The process cannot access the file 'D:\a\1\s\Stack\Opc.Ua.Core\Schema\Opc.Ua.NodeSet2.xml.zip' because it is being used by another process.
  Zip NodeSet2 completed, obj\/zipnodeset2 removed.
```

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

romanett, 2024-04-10:
New Error in latest Push: 
Stack\Opc.Ua.Core\Opc.Ua.Core.csproj(100,5): Error MSB3943: Failed to zip directory "D:\a\1\s\Stack\Opc.Ua.Core\obj\zipnodeset2" to file "D:\a\1\s\Stack\Opc.Ua.Core\Schema\Opc.Ua.NodeSet2.xml.zip".  The file 'D:\a\1\s\Stack\Opc.Ua.Core\Schema\Opc.Ua.NodeSet2.xml.zip' already exists.

mregen, 2024-04-11:
looks like a race condition between the various build flavors, maybe a different temp folder can do it

PR #2659 (2024-06-28): Fix macOS build, move https client transport back to core and other maintenance, 
## Proposed changes

- change macOS 11 ci builds to newer ones, macOS 11 is deprecated
- fix Opc.Ua.Client EventSource message numbering
- bump nerdgit to 3.6.139
- include .NET Standard 2.0 builds back in Nuget packages for client and server (complex type support requires netcoreapp2.1 which is end of life)
- include .NET Framework 4.7.2 as lowest supported .NET Framework platform
- move https client channel back to core library, as it requires only netstandard2.0 and has no other dependencies. 
- Opc.Ua.Https i [... cut]
PR #2930 (2025-01-16): Fix race condition on Nuget build
## Proposed changes

Avoid race using separate zip directories.

## Related Issues

- Fixes #2568 

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fill these out after creating the PR._

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancement (non-breaking change which adds functionality)
- [ ] Test enhancement (non-breaking change to increase test coverage)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected, requires ver [... cut]
closed 2024-06-28 by mregen