OPCFoundation/UA-.NETStandard #2585: <title>OPC UA Client

sivaprashanth, 2024-04-11:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

I am not sure why I am getting this exception when setting up the opc ua client configuration: Opc.Ua.Utils.Format(String text, Object[] args). For setting up the project i added the dll reference instead of downloading the nuget packages from the manager
  
config.ApplicationUri = Utils.Format(@"urn:{0}:" + ApplicationName + "", ServerAddress);

I also tried adding the nuget packages offline but i can't download in the visual studio offline when using this file from online : opcfoundation.netstandard.opc.ua.1.4.372.56.nupkg

### Environment

```markdown
- OS:Windows 10
- Environment:
- Runtime: v4.6.1
- Nuget Version:1.4.372.56
- Component:Opc.ua.core
- Server: kepserver
- Client: visual studio application
```

mregen, 2024-04-18:
Hi @sivaprashanth, which exception fo you get?

EthanChangAED, 2024-07-05:
Closing the issue due to lack of feedback.

closed 2024-07-05 by EthanChangAED