OPCFoundation/UA-.NETStandard #2586: GDS Server (Client) FinishRequest fails for special SubjectName

KircMax, 2024-04-16:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [X] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

When I try to have a certificate signed with a subject name like:
"\"ValueWith/Inside\""
or
"\"ValueWith=Inside\""
the GDS Server signing fails:
[image]

I call it like so:
var certificateRequest = PushClient.CreateSigningRequest(certGroup, appCertType, subjectName, regeneratePrivateKey, nonce);
var requestId = GdsClient.StartSigningRequest(appId, null, null, certificateRequest);
var certificate = GdsClient.FinishRequest(appId, requestId, out byte[] privateKey, out byte[][] issuerCertificates);

I read the spec. that this is the correct way to provide a value that contains a '/' or a '=' - please correct me if I maybe misread it:
"If the value contains a ‘/’ or a ‘=’ then it shall be enclosed in double quotes (‘”’)."
https://reference.opcfoundation.org/GDS/v105/docs/7.9.4

### Expected Behavior

works fine, certificate is signed.

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

romanett, 2024-04-20:
Hi @KircMax I could reproduce the Issue when changing the Subject Name in the Test Set of the GDS to "CN={0},DC={1},O=OPC\"=\"Foundation", or  "CN={0},DC={1},O=OPC\"/\"Foundation".
The only Method the UA .NET Stack does really use the provided subject Name is in the "StartNewKeyPairRequest" as the other methods take an existing cert request and our Server does not support changing the subject name to a value different than specified in the configuration.

However I am not really shure to which value the X500 Distinguished Name should be parsed, when provided in the format above.

The method doing the parsing you can find here:

https://github.com/OPCFoundation/UA-.NETStandard/blob/884ddf38b2b0bcef302b1fd5628140f2afc2b056/Libraries/Opc.Ua.Gds.Server.Common/ApplicationsNodeManager.cs#L750

Our parsing does not remove the '"' from the provided Subject Name, so this is the Result: 
"CN=UA Yellow Green Server,DC=dogblueberry,O="OPC"="Foundation"

which is not a valid subject name, which makes the "Finish Request" Method fail.

How does your server parse the Subject Name when provided with characters =  or / ?

OPCLabs, 2024-04-20:
@romanett, just note that per the spec, the CertificateManager may ignore the subject passed to it StartNewKeyPairRequest or StartSigningRequest (and I have seen it happen). For this reason, a Server that does not support changing the subject name in this way won't be interoperable, I think.

romanett, 2024-04-20:
In the ConfigurationNode Manager we do exactly this and ignore the provided Subject Name to keep the running server consistent with the Name specified in the xml configuration.

To me it seems we are missing a parsing step in the X509Utils.ParseDistinguishedName.

The certificate creation fails with this Exception, when an invalid subjectname is provided: WindowsCryptographicException: Die Zeichenfolge enthält einen ungültigen Schlüssel, OID, Wert oder Trennzeichen für das X500-Namensattribut (sorry for the german exception message).

Anyone has an idea how these characters shall be handled in a X500 Distinguished Name?

romanett, 2024-05-12:
@KircMax Please provide the excact subject name provided by your server in the certificate request, for us the be able to reproduce the issue

KircMax, 2024-06-10:
Hi @romanett!
I am so sorry for taking forever.
By now I can perform the CSR successfully - our server actually had some issues ...
But still I've now faced an issue with these special characters - with the '=' inside the value. 
It seems the GDS Server - upon signing it - inserts a '\' to escape the = inside the value ... 
Maybe this is already fixed by now, I've used the version 1.5.373.121... 
So certificate updating works but the Subject afterwards is not as I'd expect it ...
[image]
Here's the according certificateRequest byte[]:
[certificateRequest.txt](https://github.com/user-attachments/files/15765981/certificateRequest.txt)

romanett, 2024-07-21:
@KircMax unfortunately I can´t read out the Certificate Request bytes using:
`var pkcs10CertificationRequest = new Org.BouncyCastle.Pkcs.Pkcs10CertificationRequest(byteArray);
var info = pkcs10CertificationRequest.GetCertificationRequestInfo();
var Subjectname = info.Subject;`

So I cant find out the original subject name.

I dont see any code where we introduce "\" as escaping character manually, maybe it has something to do with the asn1 parsing.

Please reopen if you find out more.

closed 2024-07-21 by romanett