OPCFoundation/UA-.NETStandard #2587: Gds Server (Client) FinishRequest fails for ECC Server Certificate

KircMax, 2024-04-16:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [X] Help wanted

### Current Behavior

When trying to have an ECC Server Certificate signed via the GdsMaster GDS Server the FinishRequest fails:
[image]
I am calling it like this:
[image]

### Expected Behavior

certificate is issued by CA - works fine.

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

KircMax, 2024-04-16:
[provisioningcert.zip](https://github.com/OPCFoundation/UA-.NETStandard/files/14997534/provisioningcert.zip)
attached the cert. I had on the server

romanett, 2024-04-16:
did you use the newest ECC Branch from here:
https://github.com/OPCFoundation/UA-.NETStandard/pull/2398

I think the GDS is also still missing ECC support in this branch, but feel free to propose a fix adding this to the GDS using this branch as a base.

romanett, 2024-11-03:
@KircMax with my linked PR there is now basic support for ECC in the GDS. With the current implementation you need to manually specify a specify ECC CertificateType in the GDS Configuration for the DefaultCertificateGroup.

PR #2817 (2025-01-21): Extend ServerConfiguration & GDS Server for ECC
## Proposed changes

Extend ConfigurationNode Manager to support multiple Application Certificates & be able to regenerate the private Key of ECC Certificates. Add Trust List Support for User Certificates & Https Certificates.

Extend the GDS to be able to be configured with ECC Certificate Types and generate / sign ECC Certificates.
In a second step the GDS will need to be extended to allow multiple Certificate Types specified per CertificateGroup and handle multiple CAs per Certificate Group (one per Certificate Type). Also the CertificateTypes [... cut]
closed 2025-01-21 by mrsuciu