OPCFoundation/UA-.NETStandard #2590: Issue connecting to Kepware.KEPServerEX.V6

royce11co, 2024-04-19:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

I've tried using the sample code and changing the connection string to the server, but it is failing to make a connection. Is there any other settings that are needed to be changed before establishing a connection?

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

romanett, 2024-04-20:
@royce11co If you look at the configuration xml there is a log path specified, as well as the security settings. In the logs you will likely see the cause for the failed connection attempt

mregen, 2024-04-22:
@royce11co did you enable the trust on both ends (e.g. copy the cert from rejected to trusted folder)?