OPCFoundation/UA-.NETStandard #2592: Flaky build due to NodeState concurrency tests

mregen, 2024-04-22:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

Hi @saurla,
see https://opcfoundation.visualstudio.com/opcua-netstandard/_build/results?buildId=8437&view=ms.vss-test-web.build-test-results-tab&runId=1314079&resultId=110980&paneView=debug

occasionally failing, root cause yet unclear but is clearly related to the new test case which tests concurrent access to the Nodestate variables. Any idea? 

### Expected Behavior

test pass and not flaky

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

saurla, 2024-04-23:
Hi. I will take a look at the issue this week.

saurla, 2024-04-23:
I have opened a pull request which attempts to fix this issue.

PR #2594 (2024-05-01): Make NodeStateCollectionConcurrencyTests more robust
Made tests more robust by using GetConsumingEnumerable which blocks indefinitely.

## Proposed changes

These changes aim to make the NodeStateCollectionConcurrencyTests to not randomly fail in the CI build.

## Related Issues

- Fixes #2592

## Types of changes

- [ ] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancement (non-breaking change which adds functionality)
- [x] Test enhancement (non-breaking change to increase test coverage)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work a [... cut]
closed 2024-05-01 by mregen