OPCFoundation/UA-.NETStandard #2597: Something slows everything down

GurcanBicer, 2024-04-25:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

I need help with a problem.
The application is supposed to run 24/7-365 days.
Now and then, the memory usage increases, and the messages become so slow from OPC.UA.
I just got a memory dump before restarting the application.

[image]

For some reason, the subscription's IncomingMessages list has become huge.

What would be the reason for this?

### Environment

```markdown
- OS:Windows Server 2022
- Environment:Visual Studio 2022
- Runtime:.NET 4.8
- Nuget Version: 1.4.372.106
- Component: Opc.Ua.Core
- Server: Kepware
- Client: Custom Built
```

EthanChangAED, 2024-04-26:
Hello @GurcanBicer, thank you for reporting this. I noticed that you are using an earlier version of the stack. As we recently made several performance improvements, I would suggest you try with the latest release. Or wait for the April release (ETA mid-March) that would also include one more [bug fixing](https://github.com/OPCFoundation/UA-.NETStandard/pull/2540).

mregen, 2024-04-27:
The incoming message list size can be configured using `MaxMessageCount`. Could you check the setting?

GurcanBicer, 2024-04-27:
@mregen , I will try that.

The sampling rate is 250ms, and the publishing interval is 250ms.

There is only 20-25 monitored items on that subscription. And those PLC tags doesn't generate that much of a data. But if that list grows even without a value change, it is still possible, if the publishing interrupted for some reason. But I don't see any log messages regarding a publish problem.

I will upgrade the library version next week and monitor for a while.

EthanChangAED, 2024-07-05:
Resolving this issue. Feel free to reactivate if the issue still reproduces on latest version.

closed 2024-07-05 by EthanChangAED