OPCFoundation/UA-.NETStandard #2608: CertificateIdentifier:Find(bool) returns no private key after Version 1.4.367.0

sveinfolkeson, 2024-05-02:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

When I call cid.Find(true) from GlobalDiscoveryClient.cs (line 332) no certificate is returned.
This works just fine in 1.4.367.0 but not in 1.4.368 or newer.

I have noticed that there has been done some changes in the CertificateIdentifier class from ver 367 to 368 and suspect that there must be some bug in the new code.

### Steps To Reproduce

1. Start GlobalDiscoveryClient 1.4.368
2. Create certificate
3. Click on "Trust List" and then "Certificate"
4. Expected result should now be the content of the Certificate but there is nothing.

If you do the same with ver 1.4.367 you will see the generated cert.

### Environment

```markdown
- OS: Windows 11
- Environment: Visual Studio 2022
- Runtime: .NET 4.6.2
- Nuget Version: 1.4.368
- Component: GlobalDiscoveryClient->CertificateIdentifier-Find
- Server: GlobalDiscoveryServer
- Client: GlobalDiscoveryClient
```

romanett, 2024-05-02:
this recent PR allows you to specify the certificate Identifier to allow private keys:
https://github.com/OPCFoundation/UA-.NETStandard/pull/2558

You would however need to change this in the client code (library version + supply parameter "noPrivateKeys" = false),  please file a PR here if you implement this change:
https://github.com/OPCFoundation/UA-.NETStandard-Samples

sveinfolkeson, 2024-05-03:
Thank you for your fast response.
I will try this out today and give you feedback whether this solved my
problem.
-SveinF