OPCFoundation/UA-.NETStandard #2632: Programmatically creation of Macinery Nodeset on Server

MarcoOlmi, 2024-05-28:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

I am quite new to OPC UA so forgive me if the question is rather basic.
I would like to create an OPC UA Server that implements the Machinery specifications.
I've found the nodesets and succesfully compiled them with ModelCompiler.

Now i would like to create all the server structure directly in c# excluding the xml import.

Is there a trail to follow or that could help me in doing this?

The first thing i cannot understand is: do I need to create a NodeManager (inheriting NodeManager2) for the whole Machinery Nodeset or should I create more node managers and create a sort of connection between them?

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

hzeng-tfs, 2024-06-24:
To create folders and nodes manually, try to search `CreateAddressSpace`
https://github.com/search?q=repo%3AOPCFoundation%2FUA-.NETStandard-Samples%20CreateAddressSpace&type=code

for example, this one:
https://github.com/OPCFoundation/UA-.NETStandard-Samples/blob/a3e0489e3ae518c369ce7b87c4c936d4a799f87a/Workshop/Empty/Server/EmptyNodeManager.cs#L101

I am not familiar with ` Machinery specifications`, but I guess you just need one `NodeManager` which is sampler.

Hope it helps.

romanett, 2024-07-20:
Please check out the sample documentation here on how to create your own NodeManager: https://github.com/OPCFoundation/UA-.NETStandard-Samples/tree/master/Workshop/DataTypes