OPCFoundation/UA-.NETStandard #2637: Private key of application certificate must be set to exportable

ThomasNehring, 2024-06-05:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

If you import an OPC ua application certificate into the MS certificate store (into localMachine\my) and mark the key as not exportable, the application will fail to start with the error message "Key not valid for use in specified state". 
If you import the certificate with the private key exportable, the application will start normally.

### Expected Behavior

To better protect the private key, it is undesirable to force the user to make it exportable. 

### Steps To Reproduce

1. Compile the Reference Server on a Windows system
2. Set the location for the private key certificate to the MS windows certificate store:
    <ApplicationCertificate>
      <StoreType>X509Store</StoreType>
      <StorePath>LocalMachine\My</StorePath>
      <SubjectName>CN=****</SubjectName> <--! replace **** with the subject name of your certificate -->
    </ApplicationCertificate>
3. create an application with the corresponding subject and import the certificate into the MS certificate store (e.g. using the certificate snapin for mmc.exe). Ensure you set the private key to 'not exportable'
4. Try to start the reference server.

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

### Anything else?

this happens in method Opc.Ua.X509Utils.CreateCopyWithPrivateKey in X509Utils, line 514, when called from
Opc.Ua.Bindings.HttpsTransportListener.Start() in HttpsTransportListener.cs line 259. 
For a reason not clear to me it seems to be necessary to copy the certificate in this context. (?).

mregen, 2024-06-05:
Agreed, the reason for the copy is that kestrel didn't accept the cert with an ephemeral private key. For the windows cert store that extra copy may not be necessary hence the exportable may not be necessary.

romanett, 2024-06-09:
@mregen  Did the issue occur on Startup or in a later stage? Maybe we can try provoking the exception intentionally and only create the Copy if if occurs. 

Because currently Startup fails with our Auto generated Application instance Certificate if the Cert is Stored on Windows Local Machine Cert Store

PR #2761 (2024-09-12): Openstore returns null if CertificateIdentifier is empty
## Proposed changes

August testing:
- OpenStore behavior changed to return null if the storepath or storetype are invalid or empty, which is a change to the previous release. Hence some support functions should check the return value of OpenStore.
- Fix: Check for null in LoadPrivateKeyEx
- in addition: try fix the new flaky test issue
- fix #2637: an exception when the https server starts and a X509Store certificate has a non exportable private key, it is not needed to create a copy for the TLS transport. Ignore the error and start the [... cut]
commit cf2e7881 (2024-09-12): Openstore returns null if CertificateIdentifier is empty (#2761)

- OpenStore behavior changed to return null if the storepath or storetype are invalid or empty, which is a change to the previous release. Hence some support functions should check the return value of OpenStore.
- Fix: Check for null in LoadPrivateKeyEx
- in addition: try fix the new flaky test issue
- fix #2637: an exception when the https server starts and a X509Store certificate has a non exportable private key, it is not needed to create a copy for the TLS transport. Ignore the error and start the cert with the original cert [... cut]
closed 2024-09-12 by mregen