OPCFoundation/UA-.NETStandard #2638: Object with OpcUaType_NullEventNotifier gives NullReferenceException in opc ua client

kristianmo, 2024-06-05:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

I am connecting with a client to a server. The server has an Object where the event notifier is OpcUaType_Null as seen in the picture. 
[image]

This gives a NullReferenceException at line 4346 in Sessions.cs in the Opc.Ua.Client dll and aborts the Read.
[image]

### Expected Behavior

I would expect the client to detect this and continue. It could be an issue with a "bad" server also but.

### Steps To Reproduce

I assume if you create a server with a opc ua null event notifier it will happen.

### Environment

```markdown
- OS: Windows 10
- Environment: Visual studio 2022 17.10.1
- Runtime: .NET 4.6.2
- Nuget Version: 1.4.372.107
- Component: Opc.Ua.Client
- Server: Customer server
- Client: Our custom client
```

romanett, 2024-07-20:
Thank you for bringing up this issue, could you provide the fix as a Pull Request?

closed 2024-09-03 by mregen

ThomasNehring, 2024-10-08:
Do we have a server with which this can be reproduced?

Archie-Miller, 2024-10-08:
No, this was issued due to a server that did not follow the specification.

PR #2729 (2024-09-03): Check that DataValue.Value is not null when verifying the event notifier for an object
## Proposed changes

Also verify that the value of the DataValue is not null when checking the event notifier

## Related Issues

- Fixes #2638 

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fill these out after creating the PR._

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancement (non-breaking change which adds functionality)
- [ ] Test enhancement (non-breaking change to increase test coverage)
- [ ] Breaking c [... cut]
PR #2746 (2024-10-09): Client ReadNodes, throw BadInvalidType if a value type returned by an attribute is null
## Proposed changes

Handle a case where a misconfigured server returns a null value for an attribute when a value type is expected. To avoid the NullReferenceException that stops the whole ReadNodes operation, flag the Node with the service result and or return a default value if required.

## Related Issues

More comprehensive fix for
- Fixes #2729, 
- Fixes #2638 

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fill these out after c [... cut]