OPCFoundation/UA-.NETStandard #2653: Client keeps disconnecting/reconnecting to broker if it runs as publisher and subscriber at the same time

hzeng-tfs, 2024-06-21:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

I built a MQTT client application which runs as a publisher and a subscriber at the same time. 
I follow the ConsoleReferencePublisher and ConsoleReferenceSubsciber basically. I added a few `WriterGroup` and a `ReaderGroup` to the the same connection.

```
// each writer group corresponds to a MQTT topic
foreach (var topic in mDictMqttTopics)
{
    pubSubConnection.WriterGroups.Add(CreateWriterGroup(topic.Value, mqttConfig));
}

pubSubConnection.ReaderGroups.Add(CreateReaderGroup(mDictMqttSubTopics[nameof(ApplyRecipe)], mqttConfig));

//create the PubSub configuration
PubSubConfigurationDataType pubSubConfiguration = new();
pubSubConfiguration.Connections = new PubSubConnectionDataTypeCollection()
    {
        pubSubConnection
    };
```
It runs correctly with a local Mosquitto server. However, I notice that there are lots of re-connection messages on log of Mosquitto:

```
20240621T14:22:38: New connection from ::1:1667 on port 1883.
20240621T14:22:38: New client connected from ::1:1667 as ClientId_0827362928 (p5, c1, k15).
20240621T14:22:38: No will message specified.
20240621T14:22:38: Sending CONNACK to ClientId_0827362928 (0, 0)
20240621T14:22:38: New connection from ::1:1668 on port 1883.
20240621T14:22:38: Client ClientId_0827362928 already connected, closing old connection.
20240621T14:22:38: New client connected from ::1:1668 as ClientId_0827362928 (p5, c1, k15).
20240621T14:22:38: No will message specified.
20240621T14:22:38: Sending CONNACK to ClientId_0827362928 (0, 0)
20240621T14:22:38: Received SUBSCRIBE from ClientId_0827362928
20240621T14:22:38: 	Json_WriterGroup_1 (QoS 0)
20240621T14:22:38: ClientId_0827362928 0 Json_WriterGroup_1
20240621T14:22:38: Sending SUBACK to ClientId_0827362928
20240621T14:22:38: Received SUBSCRIBE from ClientId_0827362928
20240621T14:22:38: 	Json_WriterGroup_1/$Metadata (QoS 0)
20240621T14:22:38: ClientId_0827362928 0 Json_WriterGroup_1/$Metadata
20240621T14:22:38: Sending SUBACK to ClientId_0827362928
20240621T14:22:43: New connection from ::1:1670 on port 1883.
20240621T14:22:43: Client ClientId_0827362928 already connected, closing old connection.
20240621T14:22:43: New client connected from ::1:1670 as ClientId_0827362928 (p5, c1, k15).
20240621T14:22:43: No will message specified.
20240621T14:22:43: Sending CONNACK to ClientId_0827362928 (0, 0)
20240621T14:22:46: Received PUBLISH from ClientId_0827362928 (d0, q1, r0, m1, 'MySite/MyProduct_1/System', ... (395 bytes))
20240621T14:22:46: Sending PUBACK to ClientId_0827362928 (m1, rc16)
20240621T14:22:48: New connection from ::1:1672 on port 1883.
20240621T14:22:48: Client ClientId_0827362928 already connected, closing old connection.
20240621T14:22:48: New client connected from ::1:1672 as ClientId_0827362928 (p5, c1, k15).
20240621T14:22:48: No will message specified.
20240621T14:22:48: Sending CONNACK to ClientId_0827362928 (0, 0)
20240621T14:22:48: Received SUBSCRIBE from ClientId_0827362928
20240621T14:22:48: 	Json_WriterGroup_1 (QoS 0)
20240621T14:22:48: ClientId_0827362928 0 Json_WriterGroup_1
20240621T14:22:48: Sending SUBACK to ClientId_0827362928
20240621T14:22:48: Received SUBSCRIBE from ClientId_0827362928
20240621T14:22:48: 	Json_WriterGroup_1/$Metadata (QoS 0)
20240621T14:22:48: ClientId_0827362928 0 Json_WriterGroup_1/$Metadata
20240621T14:22:48: Sending SUBACK to ClientId_0827362928
20240621T14:22:53: New connection from ::1:1675 on port 1883.
[... 113 lines cut ...]
        topics).ConfigureAwait(false);
}
```

The 5s pattern on log file is caused by `private int m_reconnectIntervalSeconds = 5;` I think.

So the potential fix:
 * one `MqttClient` instance for publisher and subscriber. (mqttMultimeter use one instance https://github.com/chkr1011/mqttMultimeter/blob/main/Source/Services/Mqtt/MqttClientService.cs )
 * two `MqttClient` instances, one for each, but has different `ClientId`

### Environment

```markdown
- OS: Windows 10
- Environment:
- Runtime:
- Nuget Version: 1.5.374.70-beta PubSub
- Component: PubSub
- Server:
- Client:
```

### Anything else?

As I am still new on MQTT area, please let me know if there is an option to solve this behaviour or I did the coding wrong. Thanks.

hzeng-tfs, 2024-06-21:
I temporarily solved the issue by using two connections, one for publisher and one for subscriber to make sure they have different `ClientId`s. This confirmed my observation.

Since a connection is allowed to have a publisher and a subscriber, I think it is a bug.

by the way, it could be nice to add an option to set `ClientId` explicitly rather than using a random number only.

mrsuciu, 2024-06-28:
@hzeng-tfs Yes this is a bug, even the ClientId documentation states "Hint: This identifier needs to be unique over all used clients / devices on the broker to avoid connection issues."

PR #2672 (2024-08-05): Added PublisherMqttClientOptions and SubscriberMqttClientOptions for enabling custom configurations.
## Proposed changes

Generate different ClientIds by default for publisher and subscriber mqtt clients.
Added PublisherMqttClientOptions and SubscriberMqttClientOptions for enabling custom configurations.

## Related Issues

- Fixes #2653

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fill these out after creating the PR._

- [X] Bugfix (non-breaking change which fixes an issue)
- [X] Enhancement (non-breaking change which [... cut]
closed 2024-08-05 by mregen