OPCFoundation/UA-.NETStandard #2655: CertificateIdentifier.Find returns null despite of the fact that a certificate exists

ThomasNehring, 2024-06-24:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

This is about the method "Find" in the class "CertificateIdentifier" with the following signature:

> public static X509Certificate2 Find(X509Certificate2Collection collection, string thumbprint, string subjectName, bool needPrivateKey)

If you create a (e.g. self-signed) certificate with private key and put it and its private key into a directory certificate store with subfolders "certs" and "private", respectively, and if you then compile and run the following code snippet:

>             CertificateIdentifier certificateIdentifier = new CertificateIdentifier();
>             certificateIdentifier.StoreType = CertificateStoreType.Directory;
>             certificateIdentifier.StorePath = "<path_to_store>";
>             certificateIdentifier.Thumbprint = "thumbprint";
>             X509Certificate2 cert =
>                     await certificateIdentifier.Find(true).ConfigureAwait(false);

the value "null" is returned.

### Expected Behavior

The method should return an  X509Certificate2 instance of the certificate in the folder with private key.

### Steps To Reproduce

See the instructions in "Current Behavior".

### Environment

```markdown
- OS: Windows 11
- Environment: VS 2022
- Runtime: .Net 6
- Nuget Version: na/master branch as of 20240624 
- Component: Opc.Ua.Core
- Server:
- Client:
```

### Anything else?

The method is never executed with this set of input parameters in the (Console) ReferenceClient. This is probably the reason that this was not noticed earlier. (It seems the method is not executed at all in the (Console) Reference Client.

mrsuciu, 2024-07-09:
@ThomasNehring I am in a bit of a dilemma regarding the use case, context and expectations in which the code snippet you mention is placed.
As I understand the expectation is that anyone could execute the code snippet from any context and as a result the returned certificate must contain the private key. If this is the expectation, how would private keys protected with password be handled ? 

The behavior is as expected If the call to `Find(bool needPrivateKey)` is preceeded by a call to 'LoadPrivateKeyEx(ICertificatePasswordProvider passwordProvider)' as hinted in the code:

```
message.AppendLine("Loaded a certificate with private key from store {0}.");
message.AppendLine("Ensure to call LoadPrivateKeyEx with password provider before calling Find(true).");
```

Are you pointing towards a change of method signature to include the `ICertificatePasswordProvider ` or towards throwing an Exception in case `LoadPrivateKeyEx `hasn't been called prior to `Find(bool needPrivateKey)`?

mregen, 2024-07-11:
discussed to improve the description

mregen, 2024-09-11:
imho this is by design, because the private key may require a password..

xiaoyuvax, 2024-11-08:
met the similar problem here:
I stepped into this method and am sure the return value of certificate is not null, but after stepout the method the returning task result turns into null, don't know why!

```

 public async Task<X509Certificate2> Find(bool needPrivateKey)
 {
      ...   

     // use the single instance in the certificate cache.
     if (needPrivateKey)
     {
         certificate = m_certificate = CertificateFactory.Load(certificate, true);
     }

     return certificate;   //even when certificate!=null, after the Task returns, the result is null, don't know why!!!???
 }
```

closed 2025-02-03 by romanett