OPCFoundation/UA-.NETStandard #2657: NullReferenceException after updating to 1.5.374.36 to 1.5.374.54

DerGary, 2024-06-27:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

On Application Startup the following Exception occurs:
```
Unhandled exception. System.NullReferenceException: Object reference not set to an instance of an object.
   at Opc.Ua.Bindings.TcpTransportListener.OnAccept(Object sender, SocketAsyncEventArgs e)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Net.Sockets.SocketAsyncEventArgs.AcceptCompletionCallback(IntPtr acceptedFileDescriptor, Memory`1 socketAddress, SocketError socketError)
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading.PortableThreadPool.WorkerThread.WorkerThreadStart()
```

### Expected Behavior

No Exception occurs

### Steps To Reproduce

We now see this error after updating the libraries from 1.5.374.36 to 1.5.374.54. It also happens on the newest version, but must have to do with changes added in version .54.

For us this happens while running our unit tests, it does not seem to occur when the application is started outside of the test environment.

### Environment

```markdown
- OS: Linux WSL, Ubuntu
- Environment: VSCode / dotnet
- Runtime: .net 8
- Nuget Version: 1.5.374.54
- Component: OPCFoundation.NetStandard.Opc.Ua.Server, OPCFoundation.NetStandard.Opc.Ua.Client
- Server: StandardServer, ReverseConnectServer
- Client: Standard Session
```

### Anything else?

I think it may have something to do with this commit: [d1a9baa6c677a4e1f096ca3c902e4241fcb11256](https://github.com/OPCFoundation/UA-.NETStandard/commit/d1a9baa6c677a4e1f096ca3c902e4241fcb11256#diff-4edecb7f0fd8e1265aef4880f3f0bc8170fbc9a4d0f05e7aa4412711300efca6)

mregen, 2024-06-28:
Hi @DerGary , could you provide a line number in the source code of the OnAccept function for the crash? Maybe there is some race condition in the way you start the server in the tests.
If you configure Nuget as symbol source you should be able to get it.

DerGary, 2024-06-28:
@mregen it was a bit hard to figure out why my VSCode does not include this but it worked:

[image]

and here is even more debug information. **m_channels** is null here
[image]
[image]
[image]
[image]
[image]

mregen, 2024-06-28:
thanks, that should be sufficient to track down the issue!

PR #2661 (2024-07-01): Fix exception in TcpTransportlistener OnAccept call
## Proposed changes

- Check for m_channels null value in the OnAccept, it could be already disposed.

## Related Issues

- Fixes #2657

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fill these out after creating the PR._

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancement (non-breaking change which adds functionality)
- [ ] Test enhancement (non-breaking change to increase test coverage)
- [ ] Breaking change (fix or feature that would cause exist [... cut]
closed 2024-07-01 by mregen