OPCFoundation/UA-.NETStandard #2671: BadRequestTimeout on CoreClientUtils.SelectEndpoint after Update of Library from 1.4.371.96 to 1.5.374.36

DerGary, 2024-07-11:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

We currently have the following setup:
ARsim (B&R OPC Simulation) on Windows > WSL containing Docker > Docker Container containing OPCUA Client.

When this Code is execute with Version 1.5.374.36 :
```
var serverAddress = "opc.tcp://HOSTNAME:4841";
var endpointDescription = CoreClientUtils.SelectEndpoint(
            applicationConfiguration,
            serverAddress,
            useSecurity: true
        );
```
We obtain the following Exception:
```
Opc.Ua.ServiceResultException: BadRequestTimeout
    at Opc.Ua.Bindings.ChannelAsyncOperation`1.End(Int32 timeout, Boolean throwOnError)
    at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.EndSendRequest(IAsyncResult result)
    at Opc.Ua.DiscoveryClient.GetEndpoints(RequestHeader requestHeader, String endpointUrl, StringCollection localeIds, StringCollection profileUris, EndpointDescriptionCollection& endpoints)
    at Opc.Ua.DiscoveryClient.GetEndpoints(StringCollection profileUris)
    at Opc.Ua.Client.CoreClientUtils.SelectEndpoint(ApplicationConfiguration application, String discoveryUrl, Boolean useSecurity, Int32 discoverTimeout)
```

When Version 1.4.371.96 is used instead, it works. 

### Expected Behavior

endpointDescription is returned and no exception is raised

### Steps To Reproduce

As the Setup is somewhat complicated, it is hard to set up a sample to reproduce this.

### Environment

```markdown
- OS: Windows 10
- Environment: VSCode
- Runtime: .Net 8
- Nuget Version: 1.5.374.36
- Component: Opc.Ua.Client
- Server: B&R Simulation
```

romanett, 2024-07-21:
@DerGary I suspect some network issue here or an issue with the server. Could you please create a wireshark trace of the failed communication? If you filter to protocol type opc ua, you could find out if the client acts as defined in the spec.

perclausen, 2025-02-25:
Have the same issue trying to access a KEPServerEX on a PC connected to the same network.
If I:
- Disable IP v6 on the PC running KEPServerEX it works.
- If I use IP v4 address in the discovery URL instead of host name it works.

I have bumped from 1.4.372.56 to 1.5.374.168.

mregen, 2025-02-26:
Hi @perclausen and @DerGary, in the meantime this has become a known issue when IPv6 is enabled, some servers do not respond on the IPv6 socket. 
Since a while the tcp connect code uses the dual stack functionality built into .NET to determine if a connection with name resolution is using Ipv4 or IPv6 to connect to the server. In the past the code was handcrafted and always tried to connect via Ipv4 and Ipv6, until the first won.
For the Kepware case its a Kepware server bug. It listens and allows connections to the IPv6 socket but never sends an ack for the hello, so only after a while the connection is timing out, leading to the new behavior. Remember dual stack always prefers IPv6 if available.
The workaround for such a server is to switch the dual stack support off on the client (needs a new config etc.) or to use the known IPv4 address.

perclausen, 2025-03-04:
> Hi [@perclausen](https://github.com/perclausen) and [@DerGary](https://github.com/DerGary), in the meantime this has become a known issue when IPv6 is enabled, some servers do not respond on the [...]

Thanks for the explanation / background.

So the workaround is to set `System.Net.DisableIPv6` to true in the client's runtime configuration as outlined e.g. here:
https://devblogs.microsoft.com/dotnet/dotnet-6-networking-improvements/#an-option-to-globally-disable-ipv6

I've never encountered an automation system using IPv6, so "issue" here is that IPv6 is enabled per default in Windows + change to use DualMode sockets in the SDK. Ideally the old _handcrafted code_ could be enabled in `OPCUAClient.Config.xml` or even better opt-in on IPv6/DualMode.

marcschier, 2025-07-10:
We are not planning to re-enable the old workaround code.

closed 2025-07-10 by marcschier