OPCFoundation/UA-.NETStandard #2679: Exception "Failed to establish tcp listener sockets for Ipv4 and IPv6" after restarting server

JarJasAskom, 2024-07-19:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

Start opc ua server 
Stop opc ua server
Start opc ua server  - Exception "Failed to establish tcp listener sockets for Ipv4 and IPv6"

### Expected Behavior

Start opc ua server 
Stop opc ua server
Start opc ua server  - no errors

### Steps To Reproduce

ApplicationInstance aplication = new ApplicationInstance();
// configure application
aplication.Start()
aplication.Stop();
aplication.Start()  -> Exception

### Environment

```markdown
- OS: Windows 11
- Environment: Visual Studio 2022, Windows Forms Application
- Runtime: NET 4.8
- Nuget Version: 1.5.374.78
- Component: OPC.Ua.Core
```

closed 2024-07-19 by JarJasAskom

JarJasAskom, 2024-12-19:
The problem was not in OPC UA library but in NET 4.8 Process class. Scenario is like this:

1. Main program initializez OPC UA .
2. OPC UA opens ports.
3. Main program starts worker program using Process class.
4. Worker program iheriths handles of open ports.
5. Main program deinitailizes OPC UA and OPC UA closes ports.
6. Main program initializez again OPC UA .
7. OPC UA can't open ports because they are inherited by worker program.

Process code:

<pre>
    retVal = NativeMethods.CreateProcess (
            null,               // we don't need this since all the info is in commandLine
            commandLine,        // pointer to the command line string
            null,               // pointer to process security attributes, we don't need to inheriat the handle
            null,               // pointer to thread security attributes
            <b>true,               // handle inheritance flag</b>
            creationFlags,      // creation flags
            environmentPtr,     // pointer to new environment block
            workingDirectory,   // pointer to current directory name
            startupInfo,        // pointer to STARTUPINFO
            processInfo         // pointer to PROCESS_INFORMATION
        );     
</pre>

The solution was to create our own Process class and always set 'handle inheritance flag' to false.

zN3utr4l, 2024-12-19:
#2914

I think my problem is a little different, even though the error is the same.
My problem appears already at the first start, I open Visual Studio, Build, Start (Debug or Release), the error is shown.

I use .NET 8, I checked and no one uses the port I configured for the OPC UA Server.

zN3utr4l, 2024-12-19:
I don't know how but the error solved itself, restarted the PC, checked for updates, cleaned the solution, built again and now it works.