OPCFoundation/UA-.NETStandard #2708: Subscription wont update although it seems to be online

CryptoCodeing, 2024-08-12:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

Dear Contributers

Our APP (.NET Application implementing this library) connects to a PLC via Subscription.

Tested and it seems to work.
BUT
sadly it doesnt.

Our Customer is reporting issues daily.
The Problem could be analyze this far:
1. Subscription seems to be online and everythign seems normal
2. No Updates are provided, despite that the PLC has update the subscribed field in the OPC Server
3. Turning it on and off fixes the issue for about 30-45min.

Desperately need help on this issue and are also willing to pay for provided support.

Thanks alot for any help.

### Expected Behavior

Subscription should update everytime that the PLC provids a new value in the subscribed field 

### Steps To Reproduce

1. Install App in production enviroment
2. Start the process with Prod PLC
3. sometime the system wont carry on working, since the new values from PLC are missing (because subscription doesn't work anymore)

### Environment

```markdown
- OS:Windows Server
- Environment: 
- Runtime: .NET 
- Nuget Version:
- Component:
- Server: Windows Server
- Client: Serverside Issue (no User interface, since background process)
```

mregen, 2024-08-13:
hi @CryptoCodeing , please check with wireshark traces first if there is still publishrequest/response and the session and subscription keepalive is working. It can be many reasons.

CryptoCodeing, 2024-08-14:
@mregen 
Thanks alot for the fast response, much appreciated!
i will check and come back to you if needed.

mrsuciu, 2024-08-30:
@CryptoCodeing Any new updates on this issue ?

CryptoCodeing, 2024-08-30:
@mrsuciu  Thanks for asking.
We have found, that the issue was due to a architecture deficiency in our APP and has nothing to do with the OPC .NET Library.

closed 2024-08-31 by romanett