OPCFoundation/UA-.NETStandard #2713: Unable to reconnect a server which has changed its certificate

ccasteyde, 2024-08-14:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

If a client use the auto reconnect mechanism, it fails indefinitly to reconnect to the server if the server is shut down and reconfigured with an updated certificate, even if the endpoint use no security.
This is because Session.Open checks several security points even if no security is used, and it checks that the Create service returns the same certificate as the one stored in the endpoint. This check is fine but the endpoint is not updated by reconnection code, so it has the previous certificate.
There are similar errors if the server change its configuration, and the client should use another endpoint that the one it has used for the first connection for instance.

### Expected Behavior

One should be able at least to update the server certificate without having to restart all OPC UA client processes.

### Steps To Reproduce

Connect a client to a server.
Shut down the server.
Change the server certificate.
Restart the server.
If the client use reconnection mechanism, it won't reconnecte until it is restarted.

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

romanett, 2024-08-29:
@ccasteyde The session reconnect handler does already try to handle this case, can you try in a debugging session if this logic is triggered:
https://github.com/OPCFoundation/UA-.NETStandard/blob/6dcee7ebf1f790d34c94883ff7572852086494f2/Libraries/Opc.Ua.Client/Session/SessionReconnectHandler.cs#L422-L427

https://github.com/OPCFoundation/UA-.NETStandard/blob/6dcee7ebf1f790d34c94883ff7572852086494f2/Libraries/Opc.Ua.Client/Session/SessionReconnectHandler.cs#L467-L474

ccasteyde, 2024-08-30:
> EXPÉDITEUR EXTERNE : L'expéditeur de ce mail est externe à Codra. / EXTERNAL SENDER: The sender of this email is external to Codra.

Hello

OK, I’m using an old stack version so I do not have this code present.
That explains my problem.
I’ll update to the latest code.
Thanks

https://github.com/OPCFoundation/UA-.NETStandard/blob/6dcee7ebf1f790d34c94883ff7572852086494f2/Libraries/Opc.Ua.Client/Session/SessionReconnectHandler.cs#L422-L427

https://github.com/OPCFoundation/UA-.NETStandard/blob/6dcee7ebf1f790d34c94883ff7572852086494f2/Libraries/Opc.Ua.Client/Session/SessionReconnectHandler.cs#L467-L474

closed 2024-08-30 by romanett