OPCFoundation/UA-.NETStandard #2717: Server SDK mistakenly uses SendBufferSize as ReceiveBufferSize and vice versa

einarmo, 2024-08-17:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [X] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

When the server receives an hello message it revises its buffer sizes [here](https://github.com/OPCFoundation/UA-.NETStandard/blob/4b822fced97a235fefee89cd442479991ba31d72/Stack/Opc.Ua.Core/Stack/Tcp/TcpServerChannel.cs#L397), then sends an acknowledge message with the revised sizes.

However, according to the [standard](https://reference.opcfoundation.org/Core/Part6/v105/docs/7.1.2), the `SendBufferSize` in the acknowledge message refers to the send buffer of the _server_. I'm not entirely sure how exactly this is mixed up, but what the server SDK is doing is certainly wrong somehow, as it will send the `SendBufferSize` received in the `Hello` message back as `SendBufferSize` in the acknowledge message.

Example of incorrect behavior that I observed connecting using a different OPC-UA tool:

 - Client sends `Hello` with `ReceiveBufferSize` 8192 and `SendBufferSize` 65535.
 - Server responds with an `Acknowledge` with `ReceiveBufferSize` 8192 and `SendBufferSize` 65535.
 - This explicitly violates the standard which states in 7.1.2.4: "SendBufferSize: This value shall not be larger than the ReceiveBufferSize requested in the Hello Message".

### Expected Behavior

I'm not 100% sure as I haven't read the code too closely, but this would likely be fixed by just swapping [this line](https://github.com/OPCFoundation/UA-.NETStandard/blob/4b822fced97a235fefee89cd442479991ba31d72/Stack/Opc.Ua.Core/Stack/Tcp/TcpServerChannel.cs#L450) with the next, there may be related issues in the client SDK, though I haven't looked.

### Environment

```markdown
- OS: Fedora 40
- Environment: Command line
- Runtime: .NET 8
- Nuget Version: While investigating this I linked directly with the master branch.
- Component: Opc.Ua.Server
- Server: Reference Server
- Client: Strictly speaking: https://github.com/einarmo/opcua/tree/rewrite-master, which is a dev fork of https://github.com/locka99/opcua, running the simple client example.
```

mregen, 2024-08-17:
@einarmo thanks for the report. I can confirm your finding for the server side, its correctly done for a client hello/ack.´But some checks are missing there as well.

mregen, 2024-08-18:
@einarmo , please review #2718. I *think* the right swap is when receiving the buffer sizes, please confirm.

einarmo, 2024-08-18:
I believe you're right, looking back at the original test the server also didn't obey the requested `ReceiveBufferSize`, my client just didn't fail because it doesn't actually _have_ a receive buffer. With your fix it both sends the correct sizes in response and also only sends chunks below the requested `ReceiveBufferSize`.

PR #2718 (2024-08-19): Server SDK mistakenly uses SendBufferSize as ReceiveBufferSize and vice versa
## Proposed changes

The error was undetected as long as send and receive buffer size were the same size.
Ensure the right values are used for client and server and add some more sanity checks on buffer sizes.

## Related Issues

- Fixes #2717 

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fill these out after creating the PR._

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancement (non-breaking change which adds functionali [... cut]
closed 2024-08-19 by mregen