OPCFoundation/UA-.NETStandard #2722: Extremely outdated Dependency Microsoft.Extensions.Configuration 2.1.1 -> 8.0.0

daniatic, 2024-08-19:
### Type of issue

- [ ] Bug
- [X] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

The library uses an extremely outdated version of Microsoft.Extensions.Configuration (2.1.1) from 2018. Today there is version 8.0.0

### Expected Behavior

Keep dependencies up to date!

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

mregen, 2024-08-21:
could you provide a pointer in which project?

daniatic, 2024-08-21:
When I created this issue I just stumbled upon this outdated dependency but I did not check exactly were it came from. Now I investigated further and found that it is a transitive dependency which is introduced through `Microsoft.AspNetCore.Hosting` which itself is depracted. You're using the deprecated Microsoft.AspNetCore.* packages which are all[1] outdated and deprecated for a couple of years... Instead you should move on to the newer packages. `Microsoft.AspNetCore.Hosting` -> `Microsoft.Extensions.Hosting`.

[1] Haven't checked all of them but the majors ones are.

mregen, 2024-08-21:
Hi @daniatic, thats the https transport binding. It still supports .NET Framework and netstandard2.0 which use the deprecated kestrel servers, but the specific version we use is still supported for this purpose.
For newer .NET frameworks the https binding uses a `FrameworkReference=Microsoft.AspNetCore.App` where a newer extension package should be included.

daniatic, 2024-08-21:
I see. We are in the progress of migrating to .net 8 from .net framework 4.7.2, but this will still take some months. So then these legacy dependencies should be gone.

mregen, 2024-08-22:
@daniatic you can also exclude the https package, it is now only required for server support and was seperated out because of all the dependencies.

daniatic, 2024-08-23:
@mregen 

Thanks. You mean like so:
```
  <PackageReference Include="OPCFoundation.NetStandard.Opc.Ua">
    <Version>1.5.374.78</Version>
  </PackageReference>
  <PackageReference Include="OPCFoundation.NetStandard.Opc.Ua.Bindings.Https" Version="1.5.374.78">
    <ExcludeAssets>All</ExcludeAssets>
  </PackageReference>
```

mregen, 2024-08-24:
if the above works ok, but best is to reference only server or client packages, as needed:

```
  <PackageReference Include="OPCFoundation.NetStandard.Opc.Ua.Server">
    <Version>1.5.374.78</Version>
  </PackageReference>
```

daniatic, 2024-08-27:
Thank you! I did not know there were specific packages dedicated towards client and server. I now use the appropriate package and the outdated dependency is gone.

mregen, 2024-09-06:
Yeah, we should really deprecate the Uber-legacy package.

PR #2751 (2024-09-06): Remove https package from OPCFoundation.NetStandard.Opc.Ua dependencies, causes build issues.
## Proposed changes

Multiple isses were reported when the `OPCFoundation.NetStandard.Opc.Ua` package is used with netstandard2.0 or unity, likely caused by dependencies in the https kestrel package. Make the https package optional (user needs to add it to the list of dependencies).

Package is also marked deprecated on Nuget.org.

## Related Issues

- Fixes #2741 
- Fixes #2722 

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fil [... cut]
closed 2024-09-06 by mregen