OPCFoundation/UA-.NETStandard #2734: CTT / Reference Server: Subscription Publish implementation fails two CTT Test Cases

romanett, 2024-08-24:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [X] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

Two Subscription related CTT Test Cases fail with the Current Reference Server.
Both seem to indicate an Issue with the Publishing behaviour.
- MonitoredItem -> Monitor MinQueueSize_02 -> Test Case 9 - ModifySubscription: Revise MaxNotificationsPerPublish to 10 (from 0). DiscardOldest=FALSE.
- Subscription -> Basic ->Test Case 40 - ModifySubscription: Modify the MaxNotificationsPerPublish.

[image]

### Expected Behavior

pass all subscription related CTT Test Cases

### Steps To Reproduce

1. Start Reference Server
2. Create new CTT project
3. Copy all necessary Certs into the Trusted /Issuer Folders
4.  Start CTT Test Run with both failing test cases

### Environment

```markdown
- OS: Windows 11
- Environment: Visual Studio 2022
- Runtime: .Net 8
- Nuget Version:  1.5.374.107-preview+6dcee7ebf1
- Component: Opc.Ua.Server
- Server: Reference Server
- Client: CTT Client
```

romanett, 2024-09-08:
The tests seem to be failing due to timing problems.
No fix needed right now.

closed 2024-09-08 by romanett

codeJJL, 2024-11-11:
> The tests seem to be failing due to timing problems.
> No fix needed right now.

？What do I need to set up?