OPCFoundation/UA-.NETStandard #2739: Connecting with username/password auth no longer works

WillooWisp, 2024-08-28:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

When creating session with checkDomain set to false, and using a user identity with username and password, we get an exception. It works with no authentication however.

Opc.Ua.ServiceResultException: 'Server did not return a Certificate matching the ApplicationUri specified in the EndpointDescription.'

### Expected Behavior

With checkDomain set to false we should not get the exception 'Server did not return a Certificate matching the ApplicationUri specified in the EndpointDescription.', even if we use authentication with username/password.

### Steps To Reproduce

Use the sample reference client and connect to a server with local IP, where checkDomain set to false is required.
Using no authentication will work, but requiring username/password does not work.

### Environment

```markdown
- OS: Windows 11
- Environment: Visual Studio 2022
- Runtime: .NET 8
- Nuget Version: 1.5.374.78
- Component: N/A
- Server: N/A
- Client: Reference Client
```

### Anything else?

It works with older version, e.g. 1.5.373.121

romanett, 2024-08-28:
@WillooWisp with #2733 this check will be temporarily disabled in the August Release. 
The check domain flag is there for a different purpose and not related to the application uri matching the application uri in the certificate.
In general this message indicates an non-compliant certificate being used by the server.

closed 2024-08-28 by romanett

WillooWisp, 2024-08-28:
I see, yes the problem is not only old servers but also certificates exposed with hostname as endpoint url which will then not match the local IP when connecting.

When will this August release show up, since end of August now?

WillooWisp, 2024-08-30:
@romanett  do you know when this August release is coming out?

romanett, 2024-08-30:
@WillooWisp it will come in the next week If everything goes by plan, If you need the fix asap you can use the nuget Preview Feed linked in the Main readme.md

WillooWisp, 2024-08-30:
@romanett I just tried the latest preview version '1.5.375.71-ECC-preview' and the result is unfortunately the same, it works when no authentication is in place, but with username/password it fails with 'Server did not return a Certificate matching the ApplicationUri specified in the EndpointDescription.' and now I have checkDomain set to true again. We are connecting against a local ip for the OPC Server.

romanett, 2024-08-30:
This Version is older, this is the Preview nuget feed based on latest master: https://opcfoundation.visualstudio.com/opcua-netstandard/_packaging?_a=feed&feed=opcua-preview%40Local

WillooWisp, 2024-08-30:
Okay, strange, I had the correct feed, but it installed preview '1.5.375.71-ECC-preview' instead of '1.5.374.107-preview', since 375 build number was newer. It works with 1.5.374.107-preview though if that is the expected version?

WillooWisp, 2024-09-10:
@romanett Any news on the August release?

romanett, 2024-09-10:
@WillooWisp Release branch is already created, will come any day now after thorough testing:

https://github.com/OPCFoundation/UA-.NETStandard/tree/release/1.5.374