OPCFoundation/UA-.NETStandard #2744: <title>Deadlock in Tcp stack

cho-regin, 2024-09-02:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

I have two threads deadlocking each other; one in IdleCleanup in TcpListenerChannel.cs:110 and one in ChannelClosed in TcpTransportListener.cs:179

### Expected Behavior

no deadlock

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

mrsuciu, 2024-09-02:
@cho-regin The issues looks to be in the same area targeted by the recent fix https://github.com/OPCFoundation/UA-.NETStandard/pull/2714
Can you confirm ? If not please provide also the call stacks.

cho-regin, 2024-09-02:
The fix you mentioned looks like it would fix this issue. I've only run into this deadlock once so I don't think I can reproduce.

mregen, 2024-09-06:
supposedly fixed...thanks for reporting.

mregen, 2024-09-06:
please retest with August release

closed 2024-09-06 by mregen