OPCFoundation/UA-.NETStandard #2759: <title>CTT test:OpenSecureChannel.Response is not good; received 'BadSecurityChecksFailed (0x80130000)'. Expected: BadCertificateIssuerRevocationUnknown

codeJJL, 2024-09-10:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

Security Certificate Validation
Certificate will be validated as specified in OPC UA Part 4. This includes among others structure and signature examination. 
Allowing for some validation errors to be suppressed by administration directive.

Connect using a client certificate signed by a not trusted but known CA where there is no revocation list available.

OpenSecureChannel.Response is not good; received 'BadSecurityChecksFailed (0x80130000)'. Expected: BadCertificateIssuerRevocationUnknown 
(0x801c0000); Would accept: Good (0x00000000)

具体报错：OpenSecureChannel( MessageSecurityMode: SignAndEncrypt; RequestedSecurityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#Aes256_Sha256_RsaPss ); Result = BadSecurityChecksFailed (0x80130000) Expected: BadCertificateIssuerRevocationUnknown (0x801c0000); Would accept: Good (0x00000000)
[image]
[image]
[image]

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

mregen, 2024-09-24:
@codeJJL , please provide the CTT version. in general, the security check return values are obsfuscated to not provide an attacker insights. So BadSecurityChecksFailed is mostly a valid response. unless it is an older CTT.

romanett, 2024-11-18:
@codeJJL this seems to be implemented correctly:

https://github.com/OPCFoundation/UA-.NETStandard/blob/ce33d6387cfb8a6076c84403e42e2f67a0337295/Stack/Opc.Ua.Core/Stack/Tcp/TcpServerChannel.cs#L515-L526