OPCFoundation/UA-.NETStandard #2765: X509Store app certificate becomes unusable after a while

mregen, 2024-09-17:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

```
System.ArgumentException
  HResult=0x80070057
  Message=The CNG key handle being opened was detected to be ephemeral, but the EphemeralKey open option was not specified. (Parameter 'keyHandleOpenOptions')
  Source=System.Security.Cryptography.Cng
  StackTrace:
   at System.Security.Cryptography.CngKey.Open(SafeNCryptKeyHandle keyHandle, CngKeyHandleOpenOptions keyHandleOpenOptions)
   at Internal.Cryptography.Pal.CertificatePal.GetPrivateKey[T](Func`2 createCsp, Func`2 createCng)
   at Internal.Cryptography.Pal.CertificatePal.GetRSAPrivateKey()
   at Internal.Cryptography.Pal.CertificateExtensionsCommon.GetPrivateKey[T](X509Certificate2 certificate, Predicate`1 matchesConstraints)
   at System.Security.Cryptography.X509Certificates.RSACertificateExtensions.GetRSAPrivateKey(X509Certificate2 certificate)
   at Opc.Ua.Security.Certificates.X509PfxUtils.VerifyRSAKeyPair(X509Certificate2 certWithPublicKey, X509Certificate2 certWithPrivateKey, Boolean throwOnError) in D:\Source\Repos\UA-.NETStandard10\Libraries\Opc.Ua.Security.Certificates\X509Certificate\X509PfxUtils.cs:line 78

  This exception was originally thrown at this call stack:
    [External Code]
    Opc.Ua.Security.Certificates.X509PfxUtils.VerifyRSAKeyPair(System.Security.Cryptography.X509Certificates.X509Certificate2, System.Security.Cryptography.X509Certificates.X509Certificate2, bool) in X509PfxUtils.cs
```

### Expected Behavior

self signed certificate remains usable in X509Store

### Steps To Reproduce

- Use X509Store and CurrentUser/My as app certificate store
- Start ref server and connect with UA Expert or other clients
- Restart a few times.
- Eventually the above exception is thrown.

### Environment

```markdown
- OS: Windows 11 23H2
- Environment: VS 2022
- Runtime: .NET 8.0 / .NET Framework 4.8
- Nuget Version: 1.5.374.118
- Component: 
- Server: Ref server
- Client:
```

mregen, 2024-10-17:
so far not reproducible on all machines.

Siyuan-Xu-Wapice, 2025-03-03:
Hello, @mregen. We can reproduced the issue repetitively by following steps:
1. Config the OPC UA Server to use Windows Store path.
2. Start the UA Server.
3. Connect with a UA Client then disconnection.
4. Restart the UA server (a couples of times).

[image]

This bug really impacts the application choose to use Windows Cert Store as PKI.