OPCFoundation/UA-.NETStandard #2774: CreateSigningRequest API of spec12  ,If regeneratePrivateKey TRUE the Server shall create a new Private Key. However new private key not getting created.

charlesaugustineabb, 2024-09-26:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

CreateSigningRequest API ,If regeneratePrivateKey TRUE the Server shall create a new Private Key. However new private key not getting created.

### Expected Behavior

If regeneratePrivateKey TRUE the Server shall create a new Private Key

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

charlesaugustineabb, 2024-09-26:
with unified automation SDK it is working fine. however it is not working with foundation SDK.

mregen, 2024-09-26:
@charlesaugustineabb this feature is not implemented yet in the server.

charlesaugustineabb, 2024-09-26:
when u are planning to implement this feature? this sort extremely important feature for us . kindly let me know your plan on this?

mregen, 2024-09-26:
Assigned supposedly next release.

charlesaugustineabb, 2024-09-26:
can u please tell the  tentative date (atleast ) of the next release? will be helpfull

romanett, 2024-10-01:
@charlesaugustineabb the exact release date will depend on found bugs / regressions in the merged changes. Please check the implementation in #2778 if this has all the functionality you need.

PR #2778 (2024-10-09): [Server] GDS Push: Enable regeneratePrivatekey for CreateSigningRequest method of Server
## Proposed changes

Enable the server to generate a new application Certificate Private key when the client requests so in the CreateSigningRequest method.

## Related Issues

- Fixes #2774 

## Types of changes

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancement (non-breaking change which adds functionality)
- [ ] Test enhancement (non-breaking change to increase test coverage)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected, re [... cut]
closed 2024-10-09 by mregen