OPCFoundation/UA-.NETStandard #2783: How to create an instance node the 'right' way?

larws, 2024-10-02:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

I am currently trying to achieve the following. I am writing an OPC UA server application with UA-.NETStandard SDK and am loading my type nodes from a nodeset file. So far so good. Now during start up and runtime I would like to create instance nodes from my type nodes depending on the state of my server application.

I am using a node manager dervied from `Opc.Ua.Server.CustomNodeManager2` and found the public method `CreateNode(ServerSystemContext context, NodeId parentId, NodeId referenceTypeId, QualifiedName browseName, BaseInstanceState instance)` which sounds like what I would like to achieve. 

My code basically looks like this: 

``` csharp
BaseObjectState newDevice = new BaseObjectState(null)
{
    NodeId = new NodeId(101010, index),
    BrowseName = new QualifiedName("Demo Device", index),
    DisplayName = new Opc.Ua.LocalizedText("Demo Device"),
    TypeDefinitionId = deviceTypeId,
};

// this raises a ArgumentNullException
CreateNode(SystemContext, parentId, ReferenceTypeIds.Organizes, newDevice.BrowseName, newDevice);

```

What I am observing is a `ArgumentNullException` within `AddPredefinedNode()` where the node is assigned to the dictionary. `m_predefinedNodes[activeNode.NodeId] = activeNode;` This is caused by the fact that the `New()` method of the node manager is the standard implementation, meaning I am not overriding that behavior.

What strikes me as odd is that this doesn't work out of the box. What am I missing here? Any help would be highly appreciated.

On another note, what is the differences between `NodeState.Create` and `CustomNodeManager2.CreateNode` are they meant to solve the same problem or do the target different scenarios?

### Expected Behavior

Using an instance of `CustomNodeManager2` should work out of the box without having to override more functionality than expected.

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

ThomasNehring, 2024-11-03:
Hello larws,

sorry for the delay in responding. From what you have written it is not clear to me how you defined the types for which you want to create the instances. It would be easier if you could post a more complete description of what you have done. 
If you want to use you own types you have to provide classes which can represent these types at runtime - do you have these in place?
 
Typically this is done with the help of the model compiler which generates the types, which then have to be loaded into the server. This is done, for example, in the TestData subfolder in the Quickstarts.Servers project which is used in the Console Reference Server. Unfortunately in that example the instances are also defined through the model, so you cannot just proceed as in that case.

(I have to admit I don't know what the CreateNode method in the CustomNodeManager2 is supposed to do - it is used only in three places, as it seems).

Can you please provide some more details on what you have in place already?

BR, Thomas

larws, 2024-11-14:
Hi @ThomasNehring,
I was trying to avoid using the generated code from the model compiler and do the following instead.
* Import all the type nodes required for my server through a Nodeset, using `UANodeSet.Import` and then adding the resulting `NodeStateCollection` to the address space.
* Create instances of my loaded types by calling `CreateNode` and thus filling my address space with the needed instances.

From what I have found out so far the currently best supported way is the usage of the model compiler. In my opinion is unfortunate since the generated code is not compatible with the current versions of .net. 

Am I wrong and there is another way to achieve what I am trying to do without using the model compiler?

Thanks in advance.
BR, Stephan

romanett, 2025-01-07:
@larws what exactly do you mean with "generated code is not compatible with the current versions of .net"? The libraries rely heavily on model Compiler generated code and target .net9.

larws, 2025-01-07:
@romanett The main issue from my perspective is the missing use of nullable reference types and thus causing a bunch of warnings if NRT are enabled in the project. There is already an (open issue)[https://github.com/OPCFoundation/UA-ModelCompiler/issues/136] for the model-compiler project to address this but it seems to be of low priority.

VBA-N7, 2025-04-07:
Hi @larws i'm currently trying to develop a simple opcua server in dotnet. It's been two days i'm trying to expose a simple node but it's unsuccessful so far. I'm close to lose my mental health. In order to avoid throwing myself out of a windows, could you share a bit of your code to help me ?

romanett, 2025-04-07:
@VBA-N7 Please take a look at the sample reference server, if you did not already:https://github.com/OPCFoundation/UA-.NETStandard/tree/master/Applications/ConsoleReferenceServer

VBA-N7, 2025-04-08:
@romanett thanks your for the answer but i already did. The reference server or / and the sample or / and quickstart example are awful to read and cause a terrible headache for someone who don't have any experience in OPCUA.

Kind regards

larws, 2025-04-08:
Hi @VBA-N7, I can feel your pain. The examples can be quite overwhelming and challenging to follow. 

I am using the ModelCompiler to generate code which represents my model which I have as a NodeSet2.xml file. In the reference server that generated code can be found under [TestData.Classes.cs](https://github.com/OPCFoundation/UA-.NETStandard/blob/master/Applications/Quickstarts.Servers/TestData/TestData.Classes.cs). Once you have that code you can create an object instance node with the following steps:
* Instantiating a node in the `CreateAddressSpace` override of your `NodeManager` implementation
* Calling `Create(ISystemContext context, NodeId nodeId, QualifiedName browseName, LocalizedText displayName, bool assignNodeIds)` on the created instance
* Add the created node instance to the address space by calling `AddToAddressSpace`

If you want to do this without the model compiler code this will get messy, since the whole initialization code is quite complex and not really well documented and hard to follow. I am quite certain the mentioned steps won't solve your issue, but might give you some pointers in the right direction. Good luck ;-)

larws, 2025-04-08:
@romanett, @ThomasNehring, I think this issue can be closed since I doubt anyone is able to invest time and effort to clarify the documentation or the sample repos.

VBA-N7, 2025-04-11:
Hi again @larws,

First of all, thanks for your message. It helped a lot when I was deep-diving into the multiple examples.

I managed to get a semblance of a server running. However, I'm trying to add a node at runtime, but I'm having trouble creating a node without a parent and getting it to attach to the root.

Here's my `Program.cs`:

```csharp
using OneView_OPCUA;
using Quickstarts;

var builder = Host.CreateApplicationBuilder(args);

// Add services
builder.Services.AddHostedService<Worker>();

// Register the OPC UA Server with the DI container
builder.Services.AddSingleton<UAServer<OneViewServer>>(provider =>
{
    // Create the OPC UA server instance
    var server = new UAServer<OneViewServer>()
    {
        AutoAccept = true,
        Password = "pwd"
    };

    server.Create(Quickstarts.Utils.NodeManagerFactories);
    return server;
});

// Configure logging
builder.Logging.AddConsole();

// Allow the application to read configuration from environment variables
builder.Configuration.AddEnvironmentVariables("OPCUA_");

// Set the service name for Windows Service
var serviceName = $"{builder.Configuration.GetValue<string>("BrandName", "OneView")}_OPCUA_Server";

// Configure Windows Service support
builder.Services.AddWindowsService(options =>
{
    options.ServiceName = serviceName;
});

var app = builder.Build();
var server = app.Services.GetRequiredService<UAServer<OneViewServer>>();

await server.LoadAsync("ONEVIEW OPCUA", "OpcUaServer").ConfigureAwait(false);

app.Run();
```

When I try to add a node directly before `app.Run()`, I get the following error:

```
Exception has occurred: CLR/Opc.Ua.ServiceResultException
An exception of type 'Opc.Ua.ServiceResultException' occurred in Opc.Ua.Server.dll but was not handled in user code: 'Cannot find parent with id: i=84'
   at Opc.Ua.Server.CustomNodeManager2.CreateNode(ServerSystemContext context, NodeId parentId, NodeId referenceTypeId, QualifiedName browseName, BaseInstanceState instance) in /_/Libraries/Opc.Ua.Server/Diagnostics/CustomNodeManager.cs:line 362
```

Here is the code added that causes the error:

```csharp
await server.LoadAsync("ONEVIEW OPCUA", "OpcUaServer").ConfigureAwait(false);

ReadOnlyList<INodeManagerFactory> list = Quickstarts.Utils.NodeManagerFactories;
OneViewNodeManagerFactory bb = (OneViewNodeManagerFactory)list.FirstOrDefault(x => x is OneViewNodeManagerFactory) ?? throw new Exception("OneViewNodeManagerFactory not found in NodeManagerFactories.");

await server.CheckCertificateAsync(false).ConfigureAwait(false);
await server.StartAsync().ConfigureAwait(false);

var nodeManager = bb.Get();

// Define the dynamic node
var nodeId = new NodeId("MY_SUPER_NODE", nodeManager.NamespaceIndex);
var browseName = new QualifiedName("hello_node", nodeManager.NamespaceIndex);
var displayName = "Dynamic super node";

var variable = new BaseDataVariableState(null)
{
    NodeId = nodeId,
    BrowseName = browseName,
    DisplayName = displayName,
    DataType = Opc.Ua.DataTypeIds.Double,
    Value = 42.0,
    StatusCode = StatusCodes.Good,
    Timestamp = DateTime.UtcNow
};

// Add the node to the address space
var parentNodeId = ObjectIds.RootFolder; // Use the Objects Folder as the parent node
nodeManager.CreateNode(nodeManager.SystemContext, parentNodeId, nodeId, browseName, variable);

app.Run();
```

Could you please guide me on how to properly add a simple object to the root? 
Any insights or examples would be greatly appreciated!

Thanks in advance!

closed 2025-07-03 by marcschier