OPCFoundation/UA-.NETStandard #2787: Optimizing OPC UA Server Performance: Addressing Delay in Real-Time Updates and Alarm Generation

kais123456789, 2024-10-04:
### Type of issue

- [X] Bug
- [X] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

I'm utilizing the OPC UA bundle to set up an OPC UA server that supports both Data Access (DA) and Alarms & Events (AE). I'm reading 5,000 tags every second and exposing them through my OPC UA server sample. Additionally, I've created 5,000 alarms per second to stress-test the bundle. However, I noticed this configuration leads to memory leaks and significant delays in real-time updates and alarm generation.
After troubleshooting, I discovered that the method
public override void ReportEvent(ISystemContext context, IFilterTarget e)
takes over 5 seconds to complete. As a result, we experience delays in both real-time updates and alarm generation.
Are there any methods, configurations, or code refactoring approaches that could help resolve this delay?

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

Archie-Miller, 2024-10-22:
Currently have ConsoleReferenceClient receiving 9000 Events per second and 5000 Data notifications per second without seeing the issue described.

More details would be helpful

Environment details as requested in the Issue
Server Configuration file
Subscription count, queue sizes, sampling and publishing intervals

Thank you

romanett, 2024-12-18:
@kais123456789 please test with this branch if the issues are less noticeable (NodeManger was optimized to do less locking & MonitoredNode also should be more performant now)

https://github.com/OPCFoundation/UA-.NETStandard/pull/2895

romanett, 2025-12-03:
Please reopen if you have more information or suggestions for improvements