OPCFoundation/UA-.NETStandard #2790: Diconnecting keepalive after BadSessionNotActivated and BadSessionIdInvalid ReadResponsesHi, we faced the same issue after updating from 1.4.371.96 to 1.5.374.36

ChristianGrabmann, 2024-10-09:
Hi, we faced the same issue after updating from 1.4.371.96 to 1.5.374.36
After some time we had the same problem as [JeremyMahieu](https://github.com/JeremyMahieu) (see top)
With 1.4.371.96 we had no problems.

Does anyone have any ideas, if there were changes in the software which could cause this behaviour?

_Originally posted by @ChristianGrabmann in https://github.com/OPCFoundation/UA-.NETStandard/issues/1207#issuecomment-2385080286_

mregen, 2024-10-10:
@ChristianGrabmann, there were many improvements in the client keep alive. Have you tried to increase the SessionTimeout?

ChristianGrabmann, 2024-10-11:
Our Session Keep Alive (Client) ist set to 610000

.SetDefaultSessionTimeout(610000)
.SetMinSubscriptionLifetime(11000)

To what value should we increase?

ChristianGrabmann, 2024-10-11:
@mregen

mregen, 2024-10-11:
Hi @ChristianGrabmann, 
in general good settings for timeouts:

SessionTimeout: 30 or 60 seconds
Session KeepAlive: 5 seconds
SubscriptionLifetime: 300 seconds
SubscriptionLifetime should be > Session timeout, or subscriptions may disappear even if the session remains connected.
A session can reconnect to subscriptions, so the subscription lifetime can be considerably larger.

ChristianGrabmann, 2024-10-11:
Ok, Thanks.
That means, that our subscription Lifetime is wrong, shoul be changed to 300 Seconds and Session Timeout should get changed to 60 Seconds. We will try that.