OPCFoundation/UA-.NETStandard #2803: Provide all NuGet packages as single package download

lg2de, 2024-10-17:
### Type of issue

- [ ] Bug
- [X] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

The NuGet packages are available at nuget.org.
They can be downloaded individally

### Expected Behavior

All NuGet packages are available as an asset of the github release, e.g. as single ZIP archive.

### Anything else?

Why?
We have an internal NuGet feed with packages approved for our development.
Every time you provide a new release, I have to download manually all individual packages from nuget.org to put them into our internal feed.
I would be very efficient to download a single ZIP only.

mregen, 2024-10-17:
Hi @lg2de, do you have access to these build artifacts?
https://opcfoundation.visualstudio.com/opcua-netstandard/_build/results?buildId=9450&view=artifacts&pathAsName=false&type=publishedArtifacts

lg2de, 2024-10-17:
Yes, I have. 
But, this way will only work until the build is deleted after retention, isn't?

mregen, 2024-10-17:
looks like released artifacts are retained forever

closed 2024-10-18 by mregen

lg2de, 2024-10-18:
I compared as an example the file `opcfoundation.netstandard.opc.ua.bindings.https.1.5.374.126.nupkg` downloaded from nuget.org with the same file from the build for[ 1.5.374.126](https://opcfoundation.visualstudio.com/opcua-netstandard/_build/results?buildId=9474&view=results).
The files differs at binary level.
Looking inside I found that the file `.signature.p7s` is different.