OPCFoundation/UA-.NETStandard #2823: Failed connection due to revoked certificate with unsupported hash algorithm when using X509Store for certificates

ganko-pi, 2024-10-29:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

A `System.Security.Cryptography.CryptographicException` with message "Hash algorithm 1.2.840.113549.1.1.2 is not supported." occurs during connection to an OPC UA server when there is a certificate on a certificate revocation list (CRL) in the Windows certificate store which has an unsupported hash algorithm (in my specific case MD2 with OID 1.2.840.113549.1.1.2) even if it is not associated with OPC UA in any way. The error occurs in [Opc.Ua.X509CertificateStore in line 255](https://github.com/OPCFoundation/UA-.NETStandard/blob/release/1.5.374/Stack/Opc.Ua.Core/Security/Certificates/X509CertificateStore/X509CertificateStore.cs#L255) because the `IssuerName` could not be extracted due to the unknown hash.

### Expected Behavior

The `CryptographicException` should not fail the connection to the OPC UA server. If the certificate really belongs to the OPC UA connection the connection fails at a later point anyway.

### Steps To Reproduce

1. Operating system: Microsoft Windows 10
2. Tested with commit _0b23e5f3_ on branch _release/1.5.374_
3. Have a certificate with an invalid hash (example _MD2_) in the CRL of _Intermediate Certification Authorities_ in the Windows certificate store (I was not able to find a way to create a certificate with an invalid hash and add it to a revocation list so I cannot give instructions on this)
4. Clone the _UA-.NETStandard_ repository from GitHub (https://github.com/OPCFoundation/UA-.NETStandard)
5. Create a new C# console project with the name _OpcUaExample_ and .NET 8
6. Add the project _Opc.Ua.Client.csproj_ from _UA-.NETStandard/Libraries/Opc.Ua.Client_ and _Opc.Ua.Core.csproj_ from _UA-.NETStandard/Stack/Opc.Ua.Core_ to the solution
7. Add a project reference to _Opc.Ua.Client_ to _OpcUaExample_
8. Replace the contents of _Program.cs_ with the following:
```cs
using Opc.Ua;
using Opc.Ua.Client;
using Opc.Ua.Configuration;
using ISession = Opc.Ua.Client.ISession;

namespace OpcUaExample;

/// <summary>
/// Class containing the entry point of the program.
/// </summary>
public class Program
{

    /// <summary>
    /// Entry point of the program.
    /// </summary>
    /// <returns>A <see cref="Task"/> representing the asynchronous operation.</returns>
    public static async Task Main()
    {
        using OpcUaSessionKeeper opcUaSessionKeeper = new("opc.tcp://localhost:62541/Quickstarts/ReferenceServer");

        await opcUaSessionKeeper.ConnectOpcUaSession();
    }
}

/// <summary>
/// Class to manage a OPC UA session.
/// </summary>
public class OpcUaSessionKeeper : IDisposable
{
    private readonly string _opcUaUri;
    private ISession? _opcUaSession;
    private readonly ushort _opcUaCertificateLifetimeInMonths = 180;

    /// <summary>
    /// Constructor to instantiate an <see cref="OpcUaSessionKeeper"/>.
    /// </summary>
    public OpcUaSessionKeeper(string uri)
    {
        _opcUaUri = uri;
    }

    /// <inheritdoc/>
[... 227 lines cut ...]
      at Opc.Ua.Security.Certificates.X509CRL.Decode(Byte[] crl) in C:\source\repos\UA-.NETStandard\Libraries\Opc.Ua.Security.Certificates\X509Crl\X509Crl.cs:line 217
      at Opc.Ua.Security.Certificates.X509CRL.EnsureDecoded() in C:\source\repos\UA-.NETStandard\Libraries\Opc.Ua.Security.Certificates\X509Crl\X509Crl.cs:line 367
      at Opc.Ua.Security.Certificates.X509CRL.get_IssuerName() in C:\source\repos\UA-.NETStandard\Libraries\Opc.Ua.Security.Certificates\X509Crl\X509Crl.cs:line 102
      at Opc.Ua.X509CertificateStore.IsRevoked(X509Certificate2 issuer, X509Certificate2 certificate) in C:\source\repos\UA-.NETStandard\Stack\Opc.Ua.Core\Security\Certificates\X509CertificateStore\X509CertificateStore.cs:line 255
      at Opc.Ua.CertificateValidator.GetIssuerNoExceptionAsync(X509Certificate2 certificate, CertificateIdentifierCollection explicitList, CertificateStoreIdentifier certificateStore, Boolean checkRecovationStatus) in C:\source\repos\UA-.NETStandard\Stack\Opc.Ua.Core\Security\Certificates\CertificateValidator.cs:line 1045
      at Opc.Ua.CertificateValidator.GetIssuersNoExceptionsOnGetIssuer(X509Certificate2Collection certificates, List`1 issuers, Dictionary`2 validationErrors) in C:\source\repos\UA-.NETStandard\Stack\Opc.Ua.Core\Security\Certificates\CertificateValidator.cs:line 903
      at Opc.Ua.CertificateValidator.InternalValidateAsync(X509Certificate2Collection certificates, ConfiguredEndpoint endpoint, CancellationToken ct) in C:\source\repos\UA-.NETStandard\Stack\Opc.Ua.Core\Security\Certificates\CertificateValidator.cs:line 1145
      at Opc.Ua.CertificateValidator.ValidateAsync(X509Certificate2Collection chain, ConfiguredEndpoint endpoint, CancellationToken ct) in C:\source\repos\UA-.NETStandard\Stack\Opc.Ua.Core\Security\Certificates\CertificateValidator.cs:line 510
      at Opc.Ua.Client.Session.OpenAsync(String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales, Boolean checkDomain, CancellationToken ct) in C:\source\repos\UA-.NETStandard\Libraries\Opc.Ua.Client\Session\SessionAsync.cs:line 104
      at Opc.Ua.Client.Session.Create(ISessionInstantiator sessionInstantiator, ApplicationConfiguration configuration, ITransportWaitingConnection connection, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, Boolean checkDomain, String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales, CancellationToken ct) in C:\source\repos\UA-.NETStandard\Libraries\Opc.Ua.Client\Session\Session.cs:line 1174
```

# Suggested fix
File: https://github.com/OPCFoundation/UA-.NETStandard/blob/release/1.5.374/Stack/Opc.Ua.Core/Security/Certificates/X509CertificateStore/X509CertificateStore.cs
Line: 254 (first in `foreach`)
```cs
try
{
    X500DistinguishedName issuerName = crl.IssuerName;
}
catch (CryptographicException)
{
    continue;
}
```

romanett, 2024-10-30:
@ganko-pi thank you for the Details i will take a look, could you eventually provide the crl containing the "invalid" certificate?

ganko-pi, 2024-10-30:
Here is the crl containing the certificate with an MD2 hash. The file extension must be changed back from .crl.txt to .crl due to GitHub not allowing the upload of a file with .crl.
[cert_with_md2_hash.crl.txt](https://github.com/user-attachments/files/17567889/cert_with_md2_hash.crl.txt)

PR #2829 (2024-11-02): Improve crl handling in certificate stores
## Proposed changes

Improve crl handling in X509CertificateStore & DirectoryCertificateStore:
- allow crls with unknown hash algorithm names to sucessfully be decoded
- skip crls that fail to decode in functions enumerating over all crls of a store

## Related Issues

- Fixes #2823

## Types of changes

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancement (non-breaking change which adds functionality)
- [ ] Test enhancement (non-breaking change to increase test coverage)
- [ ] Breaking change (fix or feature that would cause exis [... cut]
closed 2024-11-03 by romanett