OPCFoundation/UA-.NETStandard #2838: <title>Setting MonitoringMode Opc.Ua.Client.MonitoredItem to MonitoringMode.Disabled does not stop receiving notifications for that particular monitored item</title>

shiladityak, 2024-11-06:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

Trying the following code,
```
var subscription = new Subscription();

MonitoredItem monItem = new MonitoriedItem((uint)1);
monItem.StartNodeId = NodeId.parse("ns=2;i=1001");
monItem.AttributeId = Attributes.Value;
monItem.SamplingInterval = -1;
monItem.DisplayName = "NodeItem1";

subscription.AddItem(monItem);
```

Now, am trying to change MonitoringMode to Disabled,
```
monItem.MonitoringMode = MonitoringMode.Disabled;
subscription.ApplyChanges();
```

Though we have changed MonitoringMode to Disabled, still we are receiving notifications for that particular monitored items from source.

### Expected Behavior

Expected behavior is that subscription should work properly and it should not receive notifications from the mentioned monitoring item.

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

romanett, 2024-12-04:
On first sight this seems like this is not the correct way to change the Monitoring Mode. There is a method available directly in the Subscription Class. 
Could you try with :
'''
subscription.SetMonitoringMode(MonitoringMode.Disabled, [1])
'''

mrsuciu, 2024-12-05:
@shiladityak Please follow @romanett's  recommendation

closed 2024-12-11 by romanett