OPCFoundation/UA-.NETStandard #2840: opc ua server unresponsive as the number of monitored items in the server increase.

vinaybr, 2024-11-12:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

We have built an OPC UA Server which receives data from an external system. The external system raises events with data values at regular intervals. when the data is received the following code is executed foreach parameter -

`node.Value = readResult.Value?.Value;`
`node.StatusCode = (uint)readResult.StatusCode;`
                                `node.Timestamp = time;`
                                `node.ClearChangeMasks(SystemContext, false);
`

As the number of parameters being monitored increases (around 7000) the server becomes unresponsive. This is observed consistently on a specific machine and not so frequently on others. Prelimnary investigation of the dump file shows many threads stuck with the below callstack -

`System.Threading.Monitor.ReliableEnter(System.Object, Boolean ByRef)`
`Opc.Ua.Server.MonitoredNode2.OnMonitoredNodeChanged(Opc.Ua.ISystemContext, Opc.Ua.NodeState, Opc.Ua.NodeStateChangeMasks)`
`Opc.Ua.NodeState.ClearChangeMasks(Opc.Ua.ISystemContext, Boolean)`

### Environment

```markdown
- OS: Win 10
- Environment: 
- Runtime: .Net6
- Nuget Version: 1.5.372.113
- Component: OPC.Ua.Server
- Server: 
- Client: UA Expert
```

romanett, 2024-12-18:
@vinaybr Please test with this Branch, locking was significantly reduced, does this solve the issue:

https://github.com/OPCFoundation/UA-.NETStandard/pull/2895

romanett, 2025-06-20:
@vinaybr Maybe you can optimize the Read Process of the EventFields to improve the runtime of the MonitoredNode2.OnMonitoredNodeChanged method.