OPCFoundation/UA-.NETStandard #2842: Security update CVE-2024-45526 detected as malware by CrowdStrike Falcon Sensor

rodolfo-chacon, 2024-11-13:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [X] Help wanted

### Current Behavior

Upgrading version from 1.5.374.36 to 1.5.374.70 is causing our antivirus software to deem our builds as malware.  We attempted upgrading to later versions, leading to the same results.

### Expected Behavior

I am expecting our builds not to be quarantined by our antivirus software.

### Environment

```markdown
- OS: Windows 10
- Environment:
- Runtime:
- Nuget Version: 1.5.374.70
- Component:
- Server:
- Client:
```

### Anything else?

Falcon Sensor-CSFalconService/Operational:

A file was quarantined because malicious behavior was detected.

romanett, 2024-11-18:
@rodolfo-chacon Does upgrading to the latest Version 1.5.374.126 solve this issue?

rodolfo-chacon, 2024-11-20:
Greetings @romanett, I've re-run all tests and still experiencing same issues with latest release 1.5.374.126.  We are trying to collect more logs for additional details.

[image]

romanett, 2024-11-28:
@rodolfo-chacon we recommend to put the wohl project/solution into a Virus Scanner excluded folder.

romanett, 2024-12-03:
@rodolfo-chacon any update? to clarify you get this message when using the nuget package to build your software and try to run this sw?

rodolfo-chacon, 2024-12-06:
Thanks @romanett for following up.  We did confirm that it was a false positive alarm and related to the interaction between installation software and crowdstrike.  Other developers ran into similar issues and were able to resolve by changing compression settings in their configurations.  Here's a link on stackoverflow discussing this issue: [Program installed with Inno Setup seen as Trojan (Wacatac.B!ml)](https://stackoverflow.com/questions/68834409/program-installed-with-inno-setup-seen-as-trojan-wacatac-bml)

closed 2024-12-06 by rodolfo-chacon