OPCFoundation/UA-.NETStandard #2860: Duplicated Log Output from TraceEventHandler and Default Logger

anhame, 2024-11-26:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

When utilizing the Utils.Tracing.TraceEventHandler for tracing, duplicate log outputs are observed. In most cases, one log entry lacks a timestamp (as expected), while the other includes a timestamp, leading to inconsistent logging behavior.
It appears that the issue is caused by RaiseTraceEvent being called twice, depending on how the log is triggered.
I would greatly appreciate any assistance.

regards
Hartmut

### Expected Behavior

Each log entry should be sent only once through the TraceEventHandler.

### Environment

```markdown
- OS:Windows 10
- Environment: Visual Studio 2022
- Runtime: .NET 4.8
- Nuget Version:
- Component: Opc.Ua.Core
- Server:
- Client:
```

### Anything else?

[image]

mregen, 2024-11-28:
Hi @anhame, the TraceEventHandler will be deprecated soon.

Please use the Microsoft Extension Ilogger interface and use the `Utils.SetLogger` API to redirect the logging events to your logging engine. See Console samples in the UA .NET Standard repo.

anhame, 2024-11-28:
Hi @mregen, thank you! Good to know. The TraceEventHandler is being used solely for migrating an existing project to ensure it functions properly.

closed 2024-11-28 by anhame