OPCFoundation/UA-.NETStandard #2862: LoadPredefinedNodes bug

xky0007, 2024-11-27:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

Load xml nodeset file has bug `{BadDecodingError 'Could not resolve namespace uri: http://opcfoundation.org/UA/2011/03/UANodeSet.xsd'}`

Inside `public class MyNodeManager : CustomNodeManager2`

`
        protected override NodeStateCollection LoadPredefinedNodes(ISystemContext context)
        {
            var importedNodeStates = new NodeStateCollection();
            FileStream fs = new FileStream("trumpf.signalbridge.ua4mt.xml", FileMode.Open);
            importedNodeStates.LoadFromXml(context, fs, true);
            return importedNodeStates;
        }
`

### Expected Behavior

Load all nodes in xml nodeset file

### Environment

```markdown
- OS:Windows 11
- Environment:VS 2022 17.11.5
- Runtime: .Net Core 6
- Nuget Version:1.4.372.76
- Component: Opc.Ua.Core
```

romanett, 2025-01-05:
@xky0007 Can you try to use the model Compiler to generate an xml for use with the LoadFromXml method.
https://github.com/OPCFoundation/UA-ModelCompiler

I suppose this will maybe solve your issue.

xky0007, 2025-01-06:
@romanett 
I followed the read.me and run `.\Opc.Ua.ModelCompiler.exe compile -d2 .\1.xml -o2 .\output` but got many files. None of these files contains the node `BrowseName="1:Running"`. 

And where can I find the difference for NodeSet and NodeSet2 definition? Thanks.

[image]

romanett, 2025-01-06:
@xky0007 i am not familiar with how to create a valid UANodeSet file.
More information can be found here: https://github.com/OPCFoundation/UA-ModelCompiler/blob/master/Opc.Ua.ModelCompiler/UA%20Model%20Design.xsd
 
NodeSet2 shall be used for any new nodesets published if i remember correctly.

You can use the abc.PredefinedNodes.xml to intialize the predefined nodes.

vj300, 2025-01-28:
I get the same error when loading a NodeSet2.xml file

`Opc.Ua.ServiceResultException
  HResult=0x80131500
  Message=Could not resolve namespace uri: http://opcfoundation.org/UA/2011/03/UANodeSet.xsd
  Source=Opc.Ua.Core
  StackTrace:
   at Opc.Ua.NodeState.LoadNode(ISystemContext context, XmlDecoder decoder) in /_/Stack/Opc.Ua.Core/Stack/State/NodeState.cs:line 1743
   at Opc.Ua.NodeStateCollection.LoadFromXml(ISystemContext context, Stream istrm, Boolean updateTables) in /_/Stack/Opc.Ua.Core/Stack/State/NodeStateCollection.cs:line 386
   at Opc.Ua.NodeStateCollection.LoadFromResource(ISystemContext context, String resourcePath, Assembly assembly, Boolean updateTables) in /_/Stack/Opc.Ua.Core/Stack/State/NodeStateCollection.cs:line 425
   at Opc.Ua.Server.CustomNodeManager2.LoadPredefinedNodes(ISystemContext context, Assembly assembly, String resourcePath, IDictionary`2 externalReferences) in /_/Libraries/Opc.Ua.Server/Diagnostics/CustomNodeManager.cs:line 512
`

larws, 2025-05-06:
@vj300 I am also running into the same issue you are describing when loading any NodeSet.xml file, for example `Opc.Ua.Machinery.NodeSet2.xml`. Could you elaborate a little more on how you worked around that? Did you remove the xml tags?

All I could achieve was to load an XML file which was generated by the model compiler from a nodeset called `xxx.PredefinedNodes.xml`. Any info would be appreciated.

vj300, 2025-05-07:
@larws , the code that i scraped together from various posts
` private void ImportNodeSet(string filePath)
 {
     var predefinedNodes = new NodeStateCollection();

     using Stream stream = new FileStream(filePath, FileMode.Open);
     var nodeSet = UANodeSet.Read(stream);
     nodeSet.Import(SystemContext, predefinedNodes);

     // Ensure that the new namespace uris are added to the server's namespace table.
     var newNamespaceUris = new List<string>();
     if (nodeSet.NamespaceUris != null && SystemContext.NamespaceUris != null)
     {
         foreach (var ns in nodeSet.NamespaceUris)
         {
             SystemContext.NamespaceUris.GetIndexOrAppend(ns);
             newNamespaceUris.Add(ns);
         }

         newNamespaceUris.AddRange(nodeSet.NamespaceUris);
     }

     // Add new namespace uris to current node manager
     if (newNamespaceUris.Count > 0)
     {
         var allNamespaceUris = new HashSet<string>(NamespaceUris);
         allNamespaceUris.UnionWith(newNamespaceUris);

         SetNamespaces(allNamespaceUris.ToArray());

         foreach (var namespaceUri in newNamespaceUris)
         {
             Server.NodeManager.RegisterNamespaceManager(namespaceUri, this);
         }
     }

     // Add new imported nodes to the predefined nodes collection
     foreach (var node in predefinedNodes)
     {
         AddPredefinedNode(SystemContext, node);
     }

     // Ensure the reverse references exist.
     AddReverseReferences(_externalReferences);
 }`

this method is in a class that derives from `CustomNodeManager2`. 
I also went through the NodeSet2.xml file and ensured that all ParentNodeId referred to a node that was defined previously in the file,  basically making sure that the parents were defined before any child nodes that referred to it. 

Probably, there is a better way to do all of this. Didn't find it, didn't have the time to go after it. If you find it out do let me know.

romanett, 2025-12-11:
Please open a new issue, If you find a Bug or have an proposal for an enhancement

closed 2025-12-11 by romanett