OPCFoundation/UA-.NETStandard #2870: Inconsistent data type for AccessRestrictions

lupokk, 2024-11-30:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

If AccessRestrictions is set to any value, an inconsistent data type error is encountered when reading with an opcua client: "Invalid Datatype: Expected OpcUaType_UInt16, Received OpcUaType_Int32"

`AccessRestrictions = AccessRestrictionType.None`

[image]

### Expected Behavior

Inglese
Il tipo di dato che esprime AccessRestrictions dovrebbe essere UInt16
The data type expressing AccessRestrictions should be UInt16

### Environment

```markdown
- OS:Windows 11
- Environment:Visual Studio 2022
- Runtime: net8
- Nuget Version: 1.5.374.126
- Component:Opc.Ua.Core
- Server:Reference Server
- Client: UA Expert
```

romanett, 2024-12-03:
@lupokk I can reproduce this issue however I am not shure yet why it occurs, AccessRestrictions support was added here: https://github.com/OPCFoundation/UA-.NETStandard/commit/b8d911c569c8bf5255c157b7ac1065ef704bb188

and all Datatypes are UInt16 / Ushort.

I will continue to investigate

lupokk, 2024-12-05:
@romanett thanks for the quick resolution.

PR #2883 (2024-12-05): Fix regression in Node AccessRestrictions: change return type from uint32 to ushort
## Proposed changes

Cast the AccessRestrictions explicitly to ushort to conform to spec (regression from 1.05.03. Nodeset update).

See also https://github.com/OPCFoundation/UA-.NETStandard/commit/1e927dd502979eebc574ca5ec22dac0937e1e9a4  

## Related Issues

- Fixes #2870

## Types of changes

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancement (non-breaking change which adds functionality)
- [ ] Test enhancement (non-breaking change to increase test coverage)
- [ ] Breaking change (fix  [... cut]
closed 2025-01-23 by romanett