OPCFoundation/UA-.NETStandard #2899: Could not load file or assembly Microsoft.AspNetCore.Hosting, Version=8.0.0.0

ktoonsez, 2024-12-10:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

I just converted my OPC server program last night from .NET Framework 4.8 to .NET 8.0 running in Windows 10/11 and every time I run my program now I keep getting the below error in the log.txt file:

12/10/2024 11:53:56.892 Could not load https Stack Listener.
 FileNotFoundException 'Could not load file or assembly 'Microsoft.AspNetCore.Hosting, Version=8.0.0.0, Culture=neutral, PublicKeyToken=adb9793829ddae60'. The system cannot find the file specified.'

There is no such thing as Microsoft.AspNetCore.Hosting version 8.0.0.0, the highest it goes is 2.2 but the only non-deprecated version 2.1.1 so what is the problem here?   When calling the server.run() it triggers the OnServerStarting(), LoadServerProperties() and CreateUserIdentityValidators() events but that is it, it stops triggering the rest of the usual events.  I deleted all the references from the OPC dll's from 4.8 version and added the ones from the net8.0 folder.  Its all the same code I have been running for a year just fine and looks the same as the sample code in the "Opc.Ua.Server.Tests" and "ReferenceServer" like nothing has changed recently.

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

romanett, 2024-12-10:
@ktoonsez please check if issue #2722 will fix your issue. If not could you please share relevant parts of your csproj file (framework, nuget dependencies) and the result of this command: dotnet list package --include-transitive

ktoonsez, 2024-12-11:
> @ktoonsez please check if issue #2722 will fix your issue. If not could you please share relevant parts of your csproj file (framework, nuget dependencies) and the result of this command: dotnet list package --include-transitive

@romanett That did the trick, I had no idea there were local versions of all those dll's besides the versions in Nuget.  Thanks.

closed 2024-12-11 by ktoonsez