OPCFoundation/UA-.NETStandard #2900: BadUserAccessDenied when the Identity token provided is not in the list of the server

KircMax, 2024-12-11:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

We throw a ServiceResultException that contains the Message 'Endpoint does not support the user identity type provided.' along with the StatusCode 'BadUserAccessDenied'.
[image]

### Expected Behavior

UAExpert e.g. return 'BadIdentityTokenInvalid'...
I would also rather expect 'BadIdentityTokenInvalid' since the provided identity token does not match to any of the ones provided by the server... 
I'm not sure what the server would answer if the client did send the request.
[image]

### Steps To Reproduce

1. disable anonymous access on the server
2. try to connect as anonymous with the client

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

romanett, 2024-12-11:
@KircMax Is sthis specifically written in the spec, as to me it seems Bad_UserAccessDenied is also fitting here (and also Bad_IdentityTokenRejected), as the requested operation ActivateSession is not allowed with the token (which is not necesarily valid, but not for the requested operation). If this is not clear maybe we should open a mantis issue

KircMax, 2024-12-11:
I didn't really find anything completely specific in the Spec...
I just found these but nowhere there it gets clear to me:
https://reference.opcfoundation.org/Core/Part4/v105/docs/7.14
[image]
https://reference.opcfoundation.org/Core/Part3/v105/docs/4.9.3
https://reference.opcfoundation.org/Core/Part4/v105/docs/7.41

But you are right, since Bad_UserAccessDenied is one possible Response in the AcitvateSession I wouldn't know when that would be sent to the client if not something like this is valid...
https://reference.opcfoundation.org/Core/Part4/v104/docs/5.6.3
[image]

PR #3012 (2025-02-28): [Server] [Client] Return BadIdentityTokenInvalid if use of anonymous identity token is not allowed
## Proposed changes

Return StatusCode.BadIdentityTokenInvalid instead of BadUserAccessDenied if the Client sends an AnonymousIdentityToken or no identity Token at all. This makes the Server pass UA CTT SessionService -> SessionBase -> TestCases -> Err-009

Make the Client throw a ServiceResultException BadIdentityTokenInvalid instead of BadUserAccessDenied to be consistent with the response a server would send. This is relevant as our client does a "pre-check" if the endpoint supports the provid [... cut]
closed 2025-02-28 by mrsuciu