OPCFoundation/UA-.NETStandard #2901: Certificate signature algorithms should be based on Curve Size in bits

mrsuciu, 2024-12-11:
### Type of issue

- [X] Bug
- [ ] Enhancement
- [X] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

The self signed ECC certificates hash algorithms for NistP384 and Brainpool384 curves use SHA256 signature instead of SHA384
as specified in profiles: [NistP384](https://profiles.opcfoundation.org/profile/2065) and [Brainpool384](https://profiles.opcfoundation.org/profile/2067)

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

romanett, 2024-12-11:
@mrsuciu don´t we already do this?
https://github.com/OPCFoundation/UA-.NETStandard/blob/c87eca918673e9f29bd770a0264a6a93f60b91b3/Stack/Opc.Ua.Core/Stack/Tcp/UaSCBinaryChannel.Symmetric.cs#L325-L343

mregen, 2024-12-11:
nope, this is about the hash used in the certificate

PR #2912 (2024-12-19): Set corresponding security signature hash and validate ECDSA security signature hash algorithm length for ECC certificates
## Proposed changes

Set corresponding security signature hash and validate security signature hash algorithm length for ECC certificates

## Related Issues

- Fixes #2901

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fill these out after creating the PR._

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancement (non-breaking change which adds functionality)
- [ ] Test enhancement ( [... cut]
closed 2024-12-19 by mrsuciu