OPCFoundation/UA-.NETStandard #2918: How to implement a method to a node that load from a xml nodeset file

xky0007, 2024-12-30:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [X] Help wanted

### Current Behavior

How to implement a method to a node that load from a xml nodeset file? 
I know the nsu and id for the method node. How to bind it to C# method?

Thanks.

romanett, 2025-01-05:
@xky0007 
To allow you to implement methods the best procedure is to generate the needed boilerplate code using the OPC Foundation Model Compiler:

https://github.com/OPCFoundation/UA-ModelCompiler

Afterwards you could override the LoadPredefined Nodes method to load the generated nodes:
https://github.com/OPCFoundation/UA-.NETStandard/blob/358337329a513ffdf76328d52b450a38b514aedd/Libraries/Opc.Ua.Gds.Server.Common/ApplicationsNodeManager.cs#L358

Now you override the AddBehaviourToPredefined Nodes method to link you implementation:

https://github.com/OPCFoundation/UA-.NETStandard/blob/358337329a513ffdf76328d52b450a38b514aedd/Libraries/Opc.Ua.Gds.Server.Common/ApplicationsNodeManager.cs#L368

Hope this helps.

xky0007, 2025-01-06:
@romanett Thanks. I override the `AddBehaviourToPredefinedNode` and try to add input args and bind to a method as follow.
But it seems the input args are not binded.
[image]

```
if(predefinedNode.DisplayName == "TestMethod2")
{
    Debug.WriteLine(predefinedNode.DisplayName);

    var node = predefinedNode as MethodState;

    node.InputArguments = new PropertyState<Argument[]>(node.Parent)
    {
        NodeId = new NodeId($"iiiii_InputArguments", 4),
        BrowseName = BrowseNames.InputArguments,
        DisplayName = new Opc.Ua.LocalizedText("Input Arguments"),
        TypeDefinitionId = VariableTypeIds.PropertyType,
        DataType = DataTypeIds.Argument,
        ValueRank = ValueRanks.OneDimension,
        Value = new Argument[]
            {
                new Argument
                {
                    Name = "Input1",
                    Description = new Opc.Ua.LocalizedText("First input parameter"),
                    DataType = DataTypeIds.String,
                    ValueRank = ValueRanks.Scalar
                },
                new Argument
                {
                    Name = "Input2",
                    Description = new Opc.Ua.LocalizedText("Second input parameter"),
                    DataType = DataTypeIds.Int32,
                    ValueRank = ValueRanks.Scalar
                }
            }
                };
    node.OnCallMethod = new GenericMethodCalledEventHandler(OnStart);

}

return base.AddBehaviourToPredefinedNode(context, predefinedNode);
```

romanett, 2025-01-07:
@xky0007 if you define the method in your  nodeset2 xml with input arguments then the modelCompiler will create an override for  MethodState with an intialization vector that represents the arguments, therefore the client will know the input / output arguments, then you can use this code to intialize:
```
MethodState passiveMethod = predefinedNode as MethodState;
if (passiveMethod == null)
{
    return predefinedNode;
}

if (passiveMethod.NodeId == MethodIds.TestMethod2)
{
    TestMethod2MethodState activeNode = new TestMethod2MethodState (passiveMethod.Parent);
    activeNode.Create(context, passiveMethod);

    // replace the node in the parent.
    if (passiveMethod.Parent != null)
    {
        passiveMethod.Parent.ReplaceChild(context, activeNode);
    }

    activeNode.OnCall = OnTestMethod2Call;

    return activeNode;
}
```