OPCFoundation/UA-.NETStandard #2934: <title>

codeJJL, 2025-01-15:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [X] Question
- [ ] Help wanted

### Current Behavior

[image]

How do I modify the values of applicationUri and productionUri, set the values I want. I modified it in the configuration file xml, but there are no values for the productionUri and applicationuri under serverConfiguration under the serverg node

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

romanett, 2025-01-15:
@codeJJL Those values are set directly on the top layer of the ApplicationConfiguration:

https://github.com/OPCFoundation/UA-.NETStandard/blob/760bfd92bce290b47496fa87046c4a5677b49603/Applications/ConsoleReferenceServer/Quickstarts.ReferenceServer.Config.xml#L9

codeJJL, 2025-01-17:
> [@codeJJL](https://github.com/codeJJL) Those values are set directly on the top layer of the ApplicationConfiguration:
> 
> [UA-.NETStandard/Applications/ConsoleReferenceServer/Quickstarts.ReferenceServer.Config.xml](https://github.com/OPCFoundation/UA-.NETStandard/blob/760bfd92bce290b47496fa87046c4a5677b49603/Applications/ConsoleReferenceServer/Quickstarts.ReferenceServer.Config.xml#L9)
> 
> Line 9 in [760bfd9](/OPCFoundation/UA-.NETStandard/commit/760bfd92bce290b47496fa87046c4a5677b49603)
> 
>  <ApplicationUri>urn:localhost:UA:Quickstarts:ReferenceServer</ApplicationUri>

[image]

[image]

[image]
I revised it again as per the reply and it didn't seem to work. I used uaexpert to check the nodeid: the values of i=25727 and i=25706 are null
This bothered me a lot, I cloned the project again and ran the source code directly, but even then, it was null. I'm sure that this approach won't change.

romanett, 2025-02-01:
@codeJJL The values are set in this method:

https://github.com/OPCFoundation/UA-.NETStandard/blob/cf828479ade7aa1f675b5f581bf84a6c3ec4b0d9/Libraries/Opc.Ua.Server/Server/ServerInternalData.cs#L661

You need to override this method like in the reference Server to set those from configuration:

https://github.com/OPCFoundation/UA-.NETStandard/blob/cf828479ade7aa1f675b5f581bf84a6c3ec4b0d9/Applications/Quickstarts.Servers/ReferenceServer/ReferenceServer.cs#L90

E.g. Like this:

protected virtual ServerProperties LoadServerProperties()
{
    return new ServerProperties {
        ProductUri = Configuration.ProductUri
    };
}

Please update if you find where the ApplicationUri is set

closed 2025-02-01 by romanett