OPCFoundation/UA-.NETStandard #2940: Restore .NET 8 LTS Support for UA-.NETStandard Stack

BoBiene, 2025-01-15:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [ ] Question
- [ ] Help wanted

### Current Behavior

With the introduction of .NET 9 support in PR #2865, .NET 8 support has been removed. This is problematic as .NET 8 is the current Long-Term Support (LTS) framework. Additionally, the .NET 9 assemblies are not backward-compatible, leaving users relying on LTS policies without a supported option.

### Expected Behavior

.NET 8 support should be retained alongside .NET 9 to ensure compatibility with LTS requirements, especially for industrial applications where LTS frameworks are mandated by organizational policies.

### Steps To Reproduce

1. Update the project to use .NET 8.
2. Attempt to use the current UA-.NETStandard stack.
3. Observe that .NET 8 is no longer supported, resulting in build failures or runtime issues.

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

perclausen, 2025-01-16:
Fully agree, below just to picture the issue with `OPCFoundation.NetStandard.Opc.Ua.Core` that references v9.0.0 of `Microsoft.Extensions.Logging.Abstractions`. For .NET 8, `OPCFoundation.NetStandard.Opc.Ua.Core` must reference v8.0.0 of `Microsoft.Extensions.Logging.Abstractions` like for .NET 6.

[image]

mregen, 2025-01-16:
According to Nuget.org, logging.abstractions 9.0 supports all other frameworks including net8.0: https://www.nuget.org/packages/Microsoft.Extensions.Logging.Abstractions/#dependencies-body-tab

perclausen, 2025-01-16:
> According to Nuget.org, logging.abstractions 9.0 supports all other frameworks including net8.0: https://www.nuget.org/packages/Microsoft.Extensions.Logging.Abstractions/#dependencies-body-tab

True, but you are still forcing users to bump `Microsoft.Extensions.Logging.Abstractions` and all related dependencies to 9.0.0.
Currently this blocks us from upgrading from 1.4.372.56 to 1.5.374.158 because we need to stick with the .NET 8 LTS versions.

Like @BoBiene mentions, we develop industrial applications where can't just bump dependencies at will.

mregen, 2025-01-20:
@perclausen @BoBiene 
Please provide feedback if the release build 1.5.374.168 in the preview channel will fix the build issues.

perclausen, 2025-01-21:
> [@perclausen](https://github.com/perclausen) [@BoBiene](https://github.com/BoBiene) Please provide feedback if the release build 1.5.374.168 in the preview channel will fix the build issues.

Build also works fine for me. Only tweak needed was to update version of `Microsoft.Extensions.Logging.Abstractions`.

mregen, 2025-01-23:
Latest minor version should be taken.

mregen, 2025-01-24:
see https://github.com/OPCFoundation/UA-.NETStandard/releases/tag/1.5.374.168

PR #2922 (2025-01-20): Some .NET 9 maintenance and JSON encoder improvements
## Proposed changes

This pull request includes various updates and improvements across multiple files. The changes focus on updating package references, enhancing debugging capabilities, and improving code efficiency and readability.

- Reduce memory allocations in JSON encoder by using stackalloc/ArrayPool 
- Fix some new warnings in .NET9 code analyzers
- Fix fuzzing error in JsonDecoder
- do not use .NET9 libraries with .NET8 LTS builds
- Bump langversion to 9 (max VS2019)
- Use official ASP.Net 2.3. libraries for .NET Framework (rel. J [... cut]