OPCFoundation/UA-.NETStandard #2955: <title>After the Nuget package is installed, the server comes with a predefined node that is a standard server. But the predefined node at the embeded level could not be found for replacement

codeJJL, 2025-01-23:
### Type of issue

- [ ] Bug
- [ ] Enhancement
- [ ] Compliance
- [x] Question
- [x] Help wanted

### Current Behavior

[image]
After the Nuget package is installed, the server comes with a predefined node that is a standard server. But the predefined node at the embeded level could not be found for replacement

### Environment

```markdown
- OS:
- Environment:
- Runtime:
- Nuget Version:
- Component:
- Server:
- Client:
```

romanett, 2025-02-01:
@codeJJL I think it will work to replace the existing server node in the CreateAdressSpace/ AddBehviourToPredefinedNodes (find it by its ObjectId) and then add your own Server Node with the embedded profile:
https://github.com/OPCFoundation/UA-.NETStandard/blob/cf828479ade7aa1f675b5f581bf84a6c3ec4b0d9/Libraries/Opc.Ua.Server/Diagnostics/DiagnosticsNodeManager.cs#L401

romanett, 2025-02-01:
@codeJJL You could use this method as a reference to implement your object

closed 2025-02-11 by romanett