OPCFoundation/UA-.NETStandard #2957: Expecting element 'ApplicationCertificate'

lukasscale, 2025-01-23:
I am getting a very strange error when wanting to run the ConsoleReferenceServer from the project.

It says:
```
OPC UA library: 1.5.374.158 @ 12/04/2024 10:31:52 -- 1.5.374.158+f5d00d934cd1c0679e5f3cc6423ae60fa2046968
Loading configuration from Quickstarts.ReferenceServer.
Load Application Configuration: Configuration file could not be loaded: /home/lukas/Workspace/Testing/UA Testserver Fancy new/ConsoleReferenceServer/bin/Debug/net9.0/Quickstarts.ReferenceServer.Config.xml
Configuration could not be loaded.
Error is: Error in line 5 position 52. 'Element' 'TrustedIssuerCertificates' from namespace 'http://opcfoundation.org/UA/SDK/Configuration.xsd' is not expected. Expecting element 'ApplicationCertificate'.
The application exits with error: Configuration file could not be loaded: /home/lukas/Workspace/Testing/UA Testserver Fancy new/ConsoleReferenceServer/bin/Debug/net9.0/Quickstarts.ReferenceServer.Config.xml
Configuration could not be loaded.
Error is: Error in line 5 position 52. 'Element' 'TrustedIssuerCertificates' from namespace 'http://opcfoundation.org/UA/SDK/Configuration.xsd' is not expected. Expecting element 'ApplicationCertificate'.
```

Obviously, this is true, because the `Quickstarts.ReferenceServer.Config.xml` in the project is slighty off from the defioned xsd.file.
The named part:
```
<SecurityConfiguration>
    <!-- Which certificate types are supported  -->
    <ApplicationCertificates>
      <CertificateIdentifier>
        <StoreType>Directory</StoreType>
        <StorePath>%LocalApplicationData%/OPC Foundation/pki/own</StorePath>
        <SubjectName>CN=Quickstart Reference Server, C=US, S=Arizona, O=OPC Foundation, DC=localhost</SubjectName>
        <CertificateTypeString>RsaSha256</CertificateTypeString>
      </CertificateIdentifier>
      <CertificateIdentifier>
        <!-- <TypeId>NistP256</TypeId> -->
        <StoreType>Directory</StoreType>
        <StorePath>%LocalApplicationData%/OPC Foundation/pki/own</StorePath>
        <SubjectName>CN=Quickstart Reference Server, C=US, S=Arizona, O=OPC Foundation, DC=localhost</SubjectName>
        <CertificateTypeString>NistP256</CertificateTypeString>
      </CertificateIdentifier>
      <CertificateIdentifier>
        <!-- <TypeId>NistP384</TypeId> -->
        <StoreType>Directory</StoreType>
        <StorePath>%LocalApplicationData%/OPC Foundation/pki/own</StorePath>
        <SubjectName>CN=Quickstart Reference Server, C=US, S=Arizona, O=OPC Foundation, DC=localhost</SubjectName>
        <CertificateTypeString>NistP384</CertificateTypeString>
      </CertificateIdentifier>
      <CertificateIdentifier>
        <!-- <TypeId>BrainpoolP256r1</TypeId> -->
        <StoreType>Directory</StoreType>
        <StorePath>%LocalApplicationData%/OPC Foundation/pki/own</StorePath>
        <SubjectName>CN=Quickstart Reference Server, C=US, S=Arizona, O=OPC Foundation, DC=localhost</SubjectName>
        <CertificateTypeString>BrainpoolP256r1</CertificateTypeString>
      </CertificateIdentifier>
      <CertificateIdentifier>
        <!-- <TypeId>BrainpoolP384r1</TypeId> -->
        <StoreType>Directory</StoreType>
        <StorePath>%LocalApplicationData%/OPC Foundation/pki/own</StorePath>
        <SubjectName>CN=Quickstart Reference Server, C=US, S=Arizona, O=OPC Foundation, DC=localhost</SubjectName>
        <CertificateTypeString>BrainpoolP384r1</CertificateTypeString>
      </CertificateIdentifier>
    </ApplicationCertificates>

    <!-- Where the other application certificates are stored -->

    <!-- Where the issuer certificate are stored (certificate authorities) -->
    <TrustedIssuerCertificates>
      <StoreType>Directory</StoreType>
      <StorePath>%LocalApplicationData%/OPC Foundation/pki/issuer</StorePath>
    </TrustedIssuerCertificates>

    <!-- Where the trust list is stored -->
    <TrustedPeerCertificates>
      <StoreType>Directory</StoreType>
      <StorePath>%LocalApplicationData%/OPC Foundation/pki/trusted</StorePath>
    </TrustedPeerCertificates>

    <!-- The directory used to store invalid certificates for later review by the administrator. -->
    <RejectedCertificateStore>
      <StoreType>Directory</StoreType>
      <StorePath>%LocalApplicationData%/OPC Foundation/pki/rejected</StorePath>
    </RejectedCertificateStore>
    <MaxRejectedCertificates>5</MaxRejectedCertificates>

    <!-- WARNING: The following setting (to automatically accept untrusted certificates) should be used
    for easy debugging purposes ONLY and turned off for production deployments! -->
    <AutoAcceptUntrustedCertificates>false</AutoAcceptUntrustedCertificates>

    <!-- WARNING: SHA1 signed certificates are by default rejected and should be phased out. 
       only nano and embedded profiles are allowed to use sha1 signed certificates. -->
    <RejectSHA1SignedCertificates>true</RejectSHA1SignedCertificates>
    <RejectUnknownRevocationStatus>true</RejectUnknownRevocationStatus>
    <MinimumCertificateKeySize>2048</MinimumCertificateKeySize>
    <AddAppCertToTrustedStore>false</AddAppCertToTrustedStore>
    <SendCertificateChain>true</SendCertificateChain>

    <!-- Where the User issuer certificates are stored -->
    <UserIssuerCertificates>
      <StoreType>Directory</StoreType>
      <StorePath>%LocalApplicationData%/OPC Foundation/pki/issuerUser</StorePath>
    </UserIssuerCertificates>

    <!-- Where the User trust list is stored-->
    <TrustedUserCertificates>
      <StoreType>Directory</StoreType>
      <StorePath>%LocalApplicationData%/OPC Foundation/pki/trustedUser</StorePath>
    </TrustedUserCertificates>
  </SecurityConfiguration>

```

It wants to have a section ApplicationCertificate without s, but has ApplicationCertificates (multiple ones).
However, if I just remove the s and choose any of them, I get the error:
```
The application exits with error: There is no cert with subject  in the configuration.
 Please generate a cert for your application,
 then copy the new cert to this location:
```
(there is no location given)

Am I doing something weirdly wrong?

I am using net9 and linux

niveditha-sooda, 2025-01-31:
When using ApplicationCertificate element without 's', it expects you to provide just the StoreType,StorePath and SubjectName without the enclosing CertificateIdentifier element. So removing the element CertificateIdentifier should make it work.

Regarding the error about loading the configuration, could you please share the config file?

You may also consider using the latest version of 1.5.374.158 which has an updated config file already in place without the ECC related configuration.

romanett, 2025-02-01:
Yes the config was updated with Version 1.5.3.375.
For previous versions please use the reference config from the corseponding release branches:
e,g,
https://github.com/OPCFoundation/UA-.NETStandard/tree/release/1.5.374

closed 2025-02-01 by romanett